realsenseai/librealsense #7731: Min Distance on L515

lyp0413, 2020-11-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | l515|
| Firmware Version                   | 01.05.00.00| 
| Operating System & Version |  Ubuntu16.04  | 
| Kernel Version (Linux Only)    |  4.15.0                                       | 
| Platform                                 | PC |
| SDK Version                            |                      | 
| Language                            | python                         | 
| Segment			| others }                          | 

### Issue Description
I can`t get depth  <0.49m from pyrealsense2 (neither 1024*768 nor 640*480),but it`s min distance is 0.25m in offical documents ,what should I do?

jagatfx, 2020-11-06:
@lyp0413 If you use adjust the preset dropdown in the RealSense Viewer to Short Range you will find that the sensor provides depth at closer ranges. You can use the button on the right to download the JSON file and inspect for the settings that change from the default configuration. In your python code you can then setup the sensor with the necessary settings or load the JSON configuration file.

Unfortunately, it looks like Advanced Mode for setting the configuration values is not yet support for the L515 in Unity as I get an error and none of the advanced settings show in the Inspector at runtime. It would be great if this was reflected in the documentation.

[image]

RealSenseSupport, 2020-12-09:
Hi @lyp0413 

Hopefully the comment helped but there are also more information around setting up and configuring the L515

https://www.intelrealsense.com/optimizing-the-lidar-camera-l515-range/
https://www.intelrealsense.com/download/13061/

We also added some new items in the latest SDK release regarding L515...this version is 2.40:
https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2400

*If we don’t hear from you in 7 days, this issue will be closed.

aidonchuk, 2020-12-14:
Hi! Same for me. Pls describe hot to set min distance to 0.20m in python.

RealSenseSupport, 2021-01-08:
Hi @lyp0413 and @aidonchuk 

Using the short range preset configures the camera for shorter range.  We have in our datasheet what we validate as short range for the camera - https://dev.intelrealsense.com/docs/lidar-camera-l515-datasheet
min distance for the camera is 0.25m

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-16 by RealSenseSupport