realsenseai/librealsense #7732: A factor in cpp

vigneshsara, 2020-11-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   { Linux (Ubuntu 14/16/17) |                                   | 
| Platform                                 | PC |
| Language                            |  {c++ }                          | 
                      | 

### Issue Description
HI 
i want to set the A factor for advance depth control in my cpp program.

auto advanced_mode_dev = dev.as<rs400::advanced_mode>();
 advanced_mode_dev.set_depth_control(0.08);

set_depth_control api  is only  available in  advanced_mode . but in this below paper mention that set_new_depth_control  api for cpp .
https://dev.intelrealsense.com/docs/white-paper-subpixel-linearity-improvement-for-intel-realsense-depth-cameras

is there set_depth_control and set_new_depth_control are same???

MartyG-RealSense, 2020-11-06:
Hi @vigneshsara   I do not have certain knowledge about this question, but as **set_new_depth_control** is only referenced in the sublinearity (A-Factor) white paper, it is reasonable to conclude that it is unique to the A-Factor and not the same as **set_depth_control** 

The A-Factor function has different names in different languages:

C:  RS2_Set_A-factor
C++: set_new_depth_control 
Python: set_amp_factor

vigneshsara, 2020-11-09:
Hi @MartyG-RealSense  
i think some confusion in paper and realsense sdk . set_amp_factor is the correct api .

code block;
// A factor setting.
   auto dev=devices_list[0];
   auto advanced_mode_dev = dev.as<rs400::advanced_mode>();
   STAFactor afactor;
   afactor.amplitude=0.08;
   advanced_mode_dev.set_amp_factor(afactor);

can you verify it and change in corresponding documentation  .

MartyG-RealSense, 2020-11-09:
I did extensive research into your question.  Curiously, it is defined as **set_amp_factor** in the API release notes when the subpixel linearity feature was launched, and also in the documentation for the SDK advanced mode functions, but not in the sublinearity white-paper.

https://intelrealsense.github.io/librealsense/doxygen/classrs400_1_1advanced__mode.html#ae60aeb0fd77af1a062abd7d1eb7ea9d0

https://github.com/IntelRealSense/librealsense/wiki/API-Changes#version-2230

I will tag this case as **Documentation** so it can be kept open and reviewed by the RealSense team members responsible for documentation.  Thanks for your report!

closed 2026-10-23 by MartyG-RealSense