realsenseai/librealsense #7734: Inconsistent Distortion Correction for Projection / Deprojection

sbillin, 2020-11-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                    | D455 | 
| SDK Version                        |  2.39.0                      | 
| Language                            |  C++                          | 

### Issue Description

It appears that distortion correction is inconsistent between the rs2_project_point_to_pixel() and rs2_deproject_pixel_to_point() functions located in rsutil.h and that the code in these functions is also inconsistent with the RealSense documentation.

### Projection in Intel RealSense SDK 2.0
https://dev.intelrealsense.com/docs/projection-in-intel-realsense-sdk-20

Let's start with the RealSense documentation:

> Distortion Models
> Based on the design of each model of RealSense device, the different streams may be exposed via different distortion models.
>
> 1. None
> An image has no distortion, as though produced by an idealized pinhole camera. This is typically the result of some hardware or software algorithm undistorting an image produced by a physical imager, but may simply indicate that the image was derived from some other image or images which were already undistorted. Images with no distortion have closed-form formulas for both projection and deprojection, and can be used with both rs2_project_point_to_pixel(...) and rs2_deproject_pixel_to_point(...).
>
> 2. Modified Brown-Conrady distortion
>An image is distorted, and has been calibrated according to a variation of the Brown-Conrady Distortion model. This model provides a closed-form formula to map from undistorted points to distorted points, while mapping in the other direction requires iteration or lookup tables. **Therefore, images with Modified Brown-Conrady Distortion are being undistorted when calling rs2_project_point_to_pixel(...)**. This model is used by the Intel RealSense D415's color image stream.
>
>3. Inverse Brown-Conrady distortion
> An image is distorted, and has been calibrated according to the inverse of the Brown-Conrady Distortion model. This model provides a closed-form formula to map from distorted points to undistored points, while mapping in the other direction requires iteration or lookup tables. **Therefore, images with Inverse Brown-Conrady Distortion are being undistorted when calling rs2_deproject_pixel_to_point(...)**. This model is used by the RealSense SR300's depth and infrared image streams.

So the take-away here is that forward distortion models, such as Brown-Conrady and Modified Brown-Conrady, support rs2_project_point_to_pixel(...) but do not work with rs2_deproject_pixel_to_point(...), while the opposite holds true for inverse distortion models, such as Inverse Brown-Conrady.

### librealsense code (rsutils.h)

Now let's take a look at how this is actually implemented (for SDK v2.39.0, the current release as of this writing), beginning with the projection function.

**rs2_project_point_to_pixel(...):** Note that the comment for this function states that it supports only "no distortion" or "forward distortion" models, which is consistent with the documentation above.
- **Problem 1**: However, instead of returning an error for the unsupported inverse distortion model, the code actually treats `RS2_DISTORTION_INVERSE_BROWN_CONRADY` as equivalent to `RS2_DISTORTION_MODIFIED_BROWN_CONRADY`:

```cpp
/* Given a point in 3D space, compute the corresponding pixel coordinates in an image with no distortion or forward distortion coefficients produced by the same camera */
static void rs2_project_point_to_pixel(float pixel[2], const struct rs2_intrinsics * intrin, const float point[3])
{
    float x = point[0] / point[2], y = point[1] / point[2];

    if ((intrin->model == RS2_DISTORTION_MODIFIED_BROWN_CONRADY) ||
        (intrin->model == RS2_DISTORTION_INVERSE_BROWN_CONRADY))
    {

        float r2  = x*x + y*y;
        float f = 1 + intrin->coeffs[0]*r2 + intrin->coeffs[1]*r2*r2 + intrin->coeffs[4]*r2*r2*r2;
        x *= f;
        y *= f;
        float dx = x + 2*intrin->coeffs[2]*x*y + intrin->coeffs[3]*(r2 + 2*x*x);
        float dy = y + 2*intrin->coeffs[3]*x*y + intrin->coeffs[2]*(r2 + 2*y*y);
        x = dx;
        y = dy;
    }

    if (intrin->model == RS2_DISTORTION_BROWN_CONRADY)
    {
        float r2 = x * x + y * y;
        float f = 1 + intrin->coeffs[0] * r2 + intrin->coeffs[1] * r2*r2 + intrin->coeffs[4] * r2*r2*r2;

        float xf = x * f;
        float yf = y * f;

        float dx = xf + 2 * intrin->coeffs[2] * x*y + intrin->coeffs[3] * (r2 + 2 * x*x);
        float dy = yf + 2 * intrin->coeffs[3] * x*y + intrin->coeffs[2] * (r2 + 2 * y*y);

        x = dx;
        y = dy;
    }
[... 87 lines cut ...]
        float r = (float)(tan(intrin->coeffs[0] * rd) / atan(2 * tan(intrin->coeffs[0] / 2.0f)));
        x *= r / rd;
        y *= r / rd;
    }

    point[0] = depth * x;
    point[1] = depth * y;
    point[2] = depth;
}
```
The discrepancies in the implementations for the RS2_DISTORTION_BROWN_CONRADY and RS2_DISTORTION_INVERSE_BROWN_CONRADY are not without consequence, as these are the two distortion models that are used by the D400 sensor.

Running `rs-enumerate-devices -c` on my D455 device shows the following intrinsic configuration:
-  depth / IR data:  **Brown Conrady**
-- But has all coefficients set to zero, which effectively provides a "no distortion" model while keeping the API limitations of Brown Conrady
-  color data:  **Inverse Brown Conrady**

Therefore, for the D455, the IR/Depth data should support rs2_project_point_to_pixel(...) whereas the color data should support rs2_deproject_pixel_to_point(...) but not vice-versa according to the documentation (unless an iterative or table lookup approach were implemented for the opposite operations, which they are not).

### Summary of Issues
**Question 1**:  For  rs2_project_point_to_pixel(...), is the equivalency of `RS2_DISTORTION_INVERSE_BROWN_CONRADY` and `RS2_DISTORTION_MODIFIED_BROWN_CONRADY` a bug?
- Note that if the inverse model was indeed supported by rs2_project_point_to_pixel(...) then you would expect it's implementation to be an inverse of its operation in rs2_deproject_pixel_to_point(...), since these two functions transform data  in different directions. However, this does not appear to be the case, which leads me to believe there is a bug here.

**Question 2**:  For rs2_deproject_pixel_to_point(...), is treating `RS2_DISTORTION_BROWN_CONRADY` as though it is effectively "no distortion" done as a hack in order to allow the D400 depth data to be processed by this function? I wonder about this because the D400 depth / IR data are effectively "no distortion" models by having all distortion coefficients equal zero.
- But if this a hack and this is the reason, wouldn't it be better to simply set the D400 intrinsic model to "no distortion", since if a sensor were to use a Brown Conrady distortion model with non-zero coefficients then this implementation would create a bug?

rafaelspring, 2020-11-07:
Looks similar to #6382

RealSenseSupport, 2020-11-09:
@rafaelspring, thank you very much for sharing the information.

RealSenseSupport, 2020-12-03:
Closing this case. Please feel free to let us know if you have other question.
Thank you.

closed 2020-12-03 by RealSenseSupport

oliver-om, 2021-04-02:
I came across this same problem with the "inverse" distortion model, and I was going to file a bug, but then found this issue (and #6382) that already raised the same questions. I was surprised to see that both issues have been closed without any action.

Looking back through the history of rsutil.h, `rs2_project_point_to_pixel` originally contained the line:
```C++
assert(intrin->model != RS2_DISTORTION_INVERSE_BROWN_CONRADY); // Cannot project to an inverse-distorted image
```
This line was commented out in #651, and then deleted in #4544. In #4544, the code for `rs2_project_point_to_pixel` was changed to treat `RS2_DISTORTION_INVERSE_BROWN_CONRADY` as a forward distortion model, even though the documentation states that it is not a forward distortion model.

The previous commenters here and in #6382 have written clear expositions of why the code seems to be wrong. It would be really helpful to get closure on whether the code or the documentation needs to be updated.