realsenseai/librealsense #7735: rs-fw-update does not work on 2.38.1

fiorano10, 2020-11-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400| 
| Firmware Version                   | 5.12.07.100 | 
| Operating System & Version |  Linux Ubuntu 18  | 
| Kernel Version (Linux Only)    |  5.3.0.53/62                                  | 
| Platform                                 | PC |
| SDK Version                            |  2.38.1                          | 
| Language                            |  C                          | 
| Segment			|  Robot                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Recently started upgrading our machines to 2.38.1 and `rs-fw-udate `no longer works. It works fine if we downgrade to 2.33.1

MartyG-RealSense, 2020-11-06:
Hi @fiorano10   I have not heard of a problem on **2.38.1** with the rs-fw-update program.  As an alternative though, could you check please whether you are able to update the firmware in the **RealSense Viewer** program?

You can do this by going to the **More** option near the top of the Viewer's side-panel, selecting the **Update Firmware** option and using the pop-up file selector box to browse to the location of a downloaded firmware .bin file.

You can also update to the firmware recommended by the SDK version that is currently installed by selecting **Install Recommended Firmware** from the same menu.   This option does not require you to provide a firmware .bin file.

[image]

MartyG-RealSense, 2020-11-13:
Hi @fiorano10   Do you still require assistance with this case, please?   Thanks!

fiorano10, 2020-11-13:
Nope, installation was messed up, thanks.

closed 2020-11-13 by fiorano10