realsenseai/librealsense #7736: L515 depth sensor only show for several seconds

alphalambdaepsilonxi, 2020-11-06:
| | |
|---|---|
|**librealsense**|2.38.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|00000000f0233221|
|**Firmware Version**|01.05.01.03|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|00000003ac2102b9|
|**Firmware Update Id**|00000003ac2102b9|

Please provide a description of the problem
After starting L500 depth sensor it will work for some seconds - than it will start to flicker and then depth info is no longer shown. The L515 works without flaws on a different computer.  

I receive the following error:

20:58:35 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\types.h:313 - hr returned: HResult 0x80004002: "Schnittstelle nicht unterstützt"
20:58:35 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\error-handling.cpp:68 - Error during polling error handler: hr returned: HResult 0x80004002: "Schnittstelle nicht unterstützt"
20:58:35 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\types.h:313 - hr returned: HResult 0x80004002: "Schnittstelle nicht unterstützt"
20:58:35 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\global_timestamp_reader.cpp:239 - Error during time_diff_keeper polling: hr returned: HResult 0x80004002: "Schnittstelle nicht unterstützt"
20:58:36 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\types.h:313 - hr returned: HResult 0x80004002: "Schnittstelle nicht unterstützt"
20:58:36 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\error-handling.cpp:68 - Error during polling error handler: hr returned: HResult 0x80004002: "Schnittstelle nicht unterstützt"
20:58:36 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\types.h:313 - hr returned: HResult 0x80004002: "Schnittstelle nicht unterstützt"

RealSenseSupport, 2020-12-23:
@alphalambdaepsilonxi 

Are you still experiencing this issue?  You mentioned that it works ok on a different computer.  Can you outline what exactly are the differences between the working computer system and the computer system that isn't working?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-13 by RealSenseSupport

alphalambdaepsilonxi, 2021-01-14:
Hi,
sorry for the delayed answer.
The issue still exists although I updated the SDK and the L515 firmware to the most current stable versions.

Hardware is perfect fine since it works great with a simple android table (10" amazon fire). An installation of only the viewer on my private PC works also perfect - everything is displayed as long as I want and in perfect quality.

Best regards
Alex