realsenseai/librealsense #7737: Firmware update failed for L515

arsalan-mousavian, 2020-11-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | N/A | 
| Operating System & Version |   Ubuntu 18.04 
| Kernel Version (Linux Only)    |  4.15.0-122                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.0      | 
| Language                            |                            | 
| Segment			| Robot                          | 

### Issue Description
I initially had problem launching the camera from ROS and then I used realsense-viewer to view the images. A firmware update option popped up and I updated it in the hope that my ROS problem goes away. Instead, in the middle of update it failed and now the camera is not accessible from realsense-viewer anymore.

The error that happened during update was the following. It's worth mentioning the camera was connected to the desktop cause I was able to see the video stream before the update.

```
 06/11 16:03:50,493 ERROR [140012377044736] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 06/11 16:03:50,493 ERROR [140012377044736] (fw-update-device.cpp:128) access failed for 8086:b55 uid: 4-6-6
 06/11 16:03:50,493 WARNING [140012377044736] (rs.cpp:306) null pointer passed for argument "device"
 06/11 16:03:50,616 ERROR [140012377044736] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 06/11 16:03:50,616 ERROR [140012377044736] (fw-update-device.cpp:128) access failed for 8086:b55 uid: 4-6-6
 06/11 16:03:50,616 WARNING [140012377044736] (rs.cpp:306) null pointer passed for argument "device"
 06/11 16:03:50,736 ERROR [140012377044736] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 06/11 16:03:50,736 ERROR [140012377044736] (fw-update-device.cpp:128) access failed for 8086:b55 uid: 4-6-6
 06/11 16:03:50,736 WARNING [140012377044736] (rs.cpp:306) null pointer passed for argument "device"
 06/11 16:03:50,857 ERROR [140012377044736] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 06/11 16:03:50,857 ERROR [140012377044736] (fw-update-device.cpp:128) access failed for 8086:b55 uid: 4-6-6
 06/11 16:03:50,857 WARNING [140012377044736] (rs.cpp:306) null pointer passed for argument "device"
 06/11 16:03:50,977 ERROR [140012377044736] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 06/11 16:03:50,977 ERROR [140012377044736] (fw-update-device.cpp:128) access failed for 8086:b55 uid: 4-6-6
 06/11 16:03:50,977 WARNING [140012377044736] (rs.cpp:306) null pointer passed for argument "device"
 06/11 16:03:51,096 ERROR [140012377044736] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 06/11 16:03:51,096 ERROR [140012377044736] (fw-update-device.cpp:128) access failed for 8086:b55 uid: 4-6-6
 06/11 16:03:51,096 WARNING [140012377044736] (rs.cpp:306) null pointer passed for argument "device"
 06/11 16:03:51,216 ERROR [140012377044736] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 06/11 16:03:51,216 ERROR [140012377044736] (fw-update-device.cpp:128) access failed for 8086:b55 uid: 4-6-6
 06/11 16:03:51,216 WARNING [140012377044736] (rs.cpp:306) null pointer passed for argument "device"
 06/11 16:03:51,337 ERROR [140012377044736] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 06/11 16:03:51,337 ERROR [140012377044736] (fw-update-device.cpp:128) access failed for 8086:b55 uid: 4-6-6
 06/11 16:03:51,337 WARNING [140012377044736] (rs.cpp:306) null pointer passed for argument "device"
 06/11 16:03:51,460 ERROR [140012377044736] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 06/11 16:03:51,460 ERROR [140012377044736] (fw-update-device.cpp:128) access failed for 8086:b55 uid: 4-6-6
 06/11 16:03:51,460 WARNING [140012377044736] (rs.cpp:306) null pointer passed for argument "device"
 06/11 16:03:51,580 ERROR [140012377044736] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 06/11 16:03:51,580 ERROR [140012377044736] (fw-update-device.cpp:128) access failed for 8086:b55 uid: 4-6-6
 06/11 16:03:51,580 WARNING [140012377044736] (rs.cpp:306) null pointer passed for argument "device"
 06/11 16:03:51,701 ERROR [140012377044736] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 06/11 16:03:51,701 ERROR [140012377044736] (fw-update-device.cpp:128) access failed for 8086:b55 uid: 4-6-6

```

I am looking for solutions to recover the camera firmware. I also disconnected and connected and relaunched realsense-viewer, it gave me update option but then it failed with the same error as above.

RealSenseSupport, 2020-12-02:
Hi @arsalan-mousavian 

Sorry for the delay here but were you able to get your camera functioning?  Do you still have the camera available to you?  In RealSense Viewer there is HW reset option, not sure if you used this on the unit.  Was the device plugged into USB3 port?

Also it would be good also if you can try the 2.40 release of librealsense and Viewer that was released.  A new FW was added there.

RealSenseSupport, 2020-12-09:
Hi @arsalan-mousavian 

Were you able to look into the comment above?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-12-16 by RealSenseSupport