realsenseai/librealsense #7739: Mark a objection in realsense coordinate.

ccooddeerr, 2020-11-07:
I just find the way to describ a thing in pointclouds coordinate,But I can't build a real coordinate that xyz can be expressed in meters。

MartyG-RealSense, 2020-11-07:
Hi @ccooddeerr   It sounds as though you want to get the 3-dimensional XYZ 'world' coordinates.  Is that correct please?

If that is correct, the links below may be helpful.

**C++**
https://community.intel.com/t5/Items-with-no-label/Getting-complete-point-cloud-from-D435/m-p/609187/highlight/true#M12699

**Python**
https://github.com/IntelRealSense/librealsense/issues/3862

MartyG-RealSense, 2020-11-17:
Hi @ccooddeerr   Do you still require assistance with this case, please?   Thanks!

ccooddeerr, 2020-11-18:
Yes, could you give some other advice?Thanks

MartyG-RealSense, 2020-11-18:
The link below provides some further resources about ball tracking to consider.

https://support.intelrealsense.com/hc/en-us/community/posts/360051277934

MartyG-RealSense, 2020-11-26:
Hi @ccooddeerr   Do you require further assistance with this case, please?   Thanks!

ccooddeerr, 2020-11-27:
Thanks for your advice ,I think its useful for my projects.

MartyG-RealSense, 2020-11-27:
Thanks very much for the update.  If you are satisfied with the outcome of this case, you can close it with the **Close Issue** button under the comment writing box.

closed 2020-12-03 by ccooddeerr