realsenseai/librealsense #7740: Import pyrealsense from pyrealsense2?

9and3, 2020-11-07:
Hello there!

Is there a way to import pyrealsense from pyrealsense2?

Let me expand. I have a code that uses some pyrealsense elements. This is an example:

`import pyrealsense as pyrs 
from pyrealsense.constants import rs_option
with pyrs.Service() as serv:
        with serv.Device() as dev:
            serial = dev.serial
            dev.wait_for_frames()
            c = dev.color
            H,W,_ = c.shape`

but I am unable to install pyrealsense (old version) through pip or from source. 
Is there a way to import the old version of pyrealsense through the new one? Or is it possible to substitute this code with more recent substitutes in pyrealsense2?

Thank you for your time and pardon my noobiness:)

MartyG-RealSense, 2020-11-07:
Hi @9and3   pyrealsense2 is not the same as the pyrealsense used with librealsense1 (the 1.12 Legacy SDK) and so that original pyrealsense and its code are not compatible with librealsense2 (also known as RealSense SDK 2.0).

Could you describe what you are trying to achieve please so that I can try to find a pyrealsense2 equivalent, please?  Thanks!

9and3, 2020-11-07:
Hello @MartyG-RealSense , thanks for your help and answer!

I am trying to run this code on Windows 10 (please find attached).
[OldCodeToUpdate.zip](https://github.com/IntelRealSense/librealsense/files/5505570/OldCodeToUpdate.zip)

More in particular, within this folder, there is this "record.py" file that would need an equivalence to pyrealsense2. Unfortunately, I am new to both pyrealsense and pyrealsense2 (PS: it's amazing!) so it's hard to find a workaround fo function like "pyrealsense.Service()" for example. It would be amazing to find a working equivalence in order to skip this vary painful process of installing pyrealsense (1.12 Legacy SDK) on my machine.

Thank you so much!

MartyG-RealSense, 2020-11-07:
I have attached the README of the project as a PDF file to this message so that others can easily see what your project does.

If you wanted to create a 3D model file, you could do so without Python by performing a 3D **point cloud** scan in the pre-made RealSense Viewer software's 3D mode and then exporting the data from the Viewer as a **ply** format file.  This can then be imported into 3D modelling software such as **MeshLab** or **Blender** and converted into a solid-mesh 3D model.

https://www.meshlab.net/

If you would prefer a Python-based approach, there is **pyntcloud**:

https://github.com/daavoo/pyntcloud

The discussion in the link below looks at a range of options for 3D reconstruction:

https://support.intelrealsense.com/hc/en-us/community/posts/360038501814-3D-Reconstruction, including the aforementioned pyntcloud:

https://support.intelrealsense.com/hc/en-us/community/posts/360038501814-3D-Reconstruction

[README.pdf](https://github.com/IntelRealSense/librealsense/files/5505596/README.pdf)

9and3, 2020-11-07:
Thank you for you answer @MartyG-RealSense ! I am familiar with the solutions you listed. I am personally using PCL but I would like to run this code because it allows to capture a complete point cloud of the object by automatically segmenting hands: this allows to scan the full body of multiple objects very quickly. I need a batch approach to my workflow.

So back to my original question: is there a way to substitute the pyrealsense part of the code "record.py" with  a pyrealsense synthax?

MartyG-RealSense, 2020-11-07:
I researched your question further, but I think that the coding work involved in doing a conversion to an equivalent in pyrealsense2 goes beyond the scope of the support that can be offered here, unfortunately.

I would recommend looking at the record.py script to analyse its general structure and see how you might be able to recreate it.  If I were working on the project myself though, I would forget about the conversion and instead google for **python hand segmentation** to see how a similar hand segmentation function might be achieved in Python by other means.

closed 2020-11-23 by MartyG-RealSense