realsenseai/librealsense #7742: D455 motion Module not streaming realsense viewer

noblegasses, 2020-11-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D455  | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version |   Linux (FriendlyDesktop)  | 
| Kernel Version (Linux Only)    |  4.4.179                                       | 
| Platform                                 | NanoPi NEO4 |
| SDK Version                            |  2.38.1                          | 
| Language                            |  C++                    | 
| Segment			|  Robot                       | 

### Issue Description
I've been trying to get the motion data from the d455, and struggling. I tried using Realsense Viewer to see if I can use that for motion data, however, turning on the motion stream doesn't seem to change the Pitch, roll and yaw of the camera regardless of rotation. Furthermore,  as well as some calibration errors. How do I resolve this?
[image]

MartyG-RealSense, 2020-11-08:
Hi @noblegasses   You should switch the RealSense Viewer into 2D mode with the **2D** option in the top corner of the Viewer to view the gyro and accel data of the Motion Module.

Are you able to view the Motion Module data successfully if you switch the Viewer to 2D mode please?

[image]

If you believe that you are getting ncorrect values for roll, pitch and yaw from the IMU, you may find the discussion in the link below useful.

https://github.com/IntelRealSense/librealsense/issues/4391

FrankCreen, 2020-11-11:
> Required Info
> Camera Model	D455
> Firmware Version	05.12.07.100
> Operating System & Version	Linux (FriendlyDesktop)
> Kernel Version (Linux Only)	4.4.179
> Platform	NanoPi NEO4
> SDK Version	2.38.1
> Language [...]

Hi ！I test the code below with a 75FPS, could you show your result? Thanks very much!

## Test Program
```
// include the librealsense C++ header file
#include <librealsense2/rs.hpp>

#include <opencv2/opencv.hpp>

#include <chrono>

using namespace std;
using namespace cv;

int main()
try
{
    //Contruct a pipeline which abstracts the device
    rs2::pipeline pipe;

    //Create a configuration for configuring the pipeline with a non default profile
    rs2::config cfg;
    //Add desired streams to configuration
    cfg.enable_stream(RS2_STREAM_COLOR, 480, 270, RS2_FORMAT_BGR8, 90);

    //Instruct pipeline to start streaming with the requested configuration
    pipe.start(cfg);

    //Sensor Settings
    auto sensor1 = pipe.get_active_profile().get_device().query_sensors()[1];
    sensor1.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);
    auto sensor2 = pipe.get_active_profile().get_device().query_sensors()[1];
    sensor2.set_option(RS2_OPTION_AUTO_EXPOSURE_PRIORITY, 0);

    // Display in a GUI
    const auto window_name = "Display Video";
    namedWindow(window_name, WINDOW_AUTOSIZE);

    //用于计时
    //show time
    auto start = chrono::system_clock::now();
    int frame = 0;

    while (waitKey(1) < 0 && getWindowProperty(window_name, WND_PROP_AUTOSIZE) >= 0)
    {
        rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera

        //获取彩色图像帧
        rs2::frame color = data.get_color_frame();

        //保存彩色图像
        Mat image(Size(480, 270), CV_8UC3, (void *)color.get_data(), Mat::AUTO_STEP);

        // Update the window with new data
        imshow(window_name, image);

        frame++;
        //one loop end
        auto end = chrono::system_clock::now();
        auto duration = chrono::duration_cast<chrono::microseconds>(end - start);
        double timecost = double(duration.count()) * chrono::microseconds::period::num / chrono::microseconds::period::den;

        cout << frame / timecost << " FPS" << endl;
    }

    return 0;
}
catch (Exception ex)
{
    cout << "Error!!!" << endl;
    cerr << ex.msg << endl;
}
```

MartyG-RealSense, 2020-11-11:
Hi @FrankCreen    Please post your question as a new issue using the **New Issue** button at the link below and do not post it on multiple D455 cases if the question is unrelated to that case.   Thank you.

https://github.com/IntelRealSense/librealsense/issues/

noblegasses, 2020-11-11:
Hi, 
thanks, the 2d view solves my issue. this might require opening a separate issue, but how do I stream this data in c++ or python? the pose stream seems to fail, as does storing the accel and gyro streams in frames.

MartyG-RealSense, 2020-11-11:
Hi @noblegasses   You could take a look at the SDK's **rs-motion** C++ example program.

https://github.com/IntelRealSense/librealsense/tree/master/examples/motion

If you plan to stream depth, color and IMU simultaneously then putting IMU on one pipeline and depth and color on the other can improve performance.  The link below has a Python example script for this:

https://github.com/IntelRealSense/librealsense/issues/5628#issuecomment-575943238

MartyG-RealSense, 2020-11-19:
Hi @noblegasses   Do you require further assistance with this case, please?   Thanks!

noblegasses, 2020-11-19:
my apologies, I don't require any further help at this time. I simply forgot to close the issue. Thanks!

closed 2020-11-19 by noblegasses

MartyG-RealSense, 2020-11-19:
No apologies needed - thanks very much for the update!