realsenseai/librealsense #7743: ARBackground sample runs slowly with USB extension cable even though realsense viewer has no problem with it.

krijnlol, 2020-11-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435                                                      |
| Firmware Version                   | 05.12.07.100                                           | 
| Operating System & Version |   Windows 10                                         | 
| Kernel Version (Linux Only)    |  does not apply then                                 | 
| Platform                                 | PC                                                          |
| SDK Version                            |  2                                                           | 
| Language                            |  unity                                                        | 
| Segment			|  AR                                                                 | 

### Issue Description
<Describe your issue / question / feature request / etc..>

alright, so I am trying to make an AR application I am using unity. so first I tried the ARBackground sample with the realsense directly connected to my pc which worked fine. but when I connected it to an USB extension cable it had a very low frame rate something like every 3 seconds even though the stream was a lot faster when connected in the same way but viewed through realsense viewer.

thanks for reading my issue if you can help then please reply hope you have a good day (:

MartyG-RealSense, 2020-11-08:
Hi @krijnlol   How long is the USB extension cable that you are using please?  And are you using the **ARBackground** sample with Unity 2019 or Unity 2020?  Thanks!

krijnlol, 2020-11-09:
> Hi @krijnlol How long is the USB extension cable that you are using please? And are you using the **ARBackground** sample with Unity 2019 or Unity 2020? Thanks!

the extension cable is 3 meters and the cable to the realsense is 1 meter so it 4 meters in total.
I'm using unity 2019.3.9f1.

MartyG-RealSense, 2020-11-09:
Thank you very much for the information.  Unity 2019.3 should be fine to use the Unity wrapper samples with.

When using your own choice of USB cable with RealSense that is longer than 2 meters, it is important to use a high quality cable or you may experience performance degradation.  This is because of the high volume of data that RealSense transmits through the cable.  You should also make sure that your extension cable is a USB 3 cable and not the slower USB 2 type.

Intel's officially recommended supplier for high quality cables for the 400 Series cameras is **Newnex**.

https://www.newnex.com/realsense-3d-camera-connectivity.php

krijnlol, 2020-11-09:
> Thank you very much for the information. Unity 2019.3 should be fine to use the Unity wrapper samples with.
>
> When using your own choice of USB cable with RealSense that is longer than 2 meters, it [...]

alright. I'm going to buy a new cable then. are you really sure that is the problem? with the same cable setup it had no problem in the intel realsense viewer.

MartyG-RealSense, 2020-11-09:
The way I understood your opening comment was that it works fine if you just connect with the 1 meter official cable supplied with the camera, but has problems if you add onto it your own choice of 3m extension cable.  This suggests that if the ARBackground sample works fine when just using the 1 m cable (and so that cable does not have problems), but then the slow-down happens when adding the 3 m own-choice cable onto the 1 m one to make a total length of 4 m.  Did I understand correctly, please?   Thanks!

krijnlol, 2020-11-09:
> The way I understood your opening comment was that it works fine if you just connect with the 1 meter official cable supplied with the camera, but has problems if you add onto it your own choice of 3m extension cable.
> 
> The ARBackground sample works fine when just using the 1 m cable (and so that cable does not have problems), but then the slow-down happens when adding the 3 m own-choice cable onto the 1 m one to make a total length of 4 m. This therefore suggests that the problem is with the 3 m cable, not both cables.
> 
> Did I understand correctly, please? Thanks!

yea that's partially correct but with both cables connected (4 meters) in the intel realsense viewer tool I don't experience any slowdown

MartyG-RealSense, 2020-11-09:
Thanks for the clarification.  Do you experience slowdown with any of the other Unity samples when using the 3 m cable extension, please?

krijnlol, 2020-11-09:
well yes only in the 2D samples the point clouds run very smooth

krijnlol, 2020-11-09:
actually, the alignment sample also runs smoothly and the background segmentation too

krijnlol, 2020-11-09:
okay so only the TexturesDepthAndColor sample and the ARBackground sample run slowly

MartyG-RealSense, 2020-11-09:
Thank you.   If these programs work fine with just the 1 m cable, I wonder if these two programs have higher processing demands than the other example programs that make it harder to maintain transmission quality over greater distance when using the 3 m extension.

krijnlol, 2020-11-09:
okay so I just discovered that when I set the fps of the devices to 15 it runs smoothly

krijnlol, 2020-11-09:
I'm going to close this issue thanks a lot for helping me (:

MartyG-RealSense, 2020-11-09:
That would seem to add weight to the theory that the 3 m cable was havig difficulty coping (perhaps due to quality issues), as reducing FPS will reduce the volume of bandwidth being transmitted through the cable.

closed 2020-11-09 by krijnlol

MartyG-RealSense, 2020-11-09:
You are very welcome   :)