realsenseai/librealsense #7744: is it possible for `pyrealsense2.device` instance to become invalid?

Chandler, 2020-11-08:
If have an instance of `pyrealsense2.device` and I call 
```
depth_sensor = dev.first_depth_sensor()
depth_sensor.set_option(rs.option.visual_preset, 3)
```
and it does not throw an exception, is it guaranteed to work? I am finding that sometimes it does not work and my settings changes do not take effect. I can only fix it by re-querying the device before making my changes.

```
....
profile = pipeline.start(config)
dev = profile.get_device()
depth_sensor = dev.first_depth_sensor()
depth_sensor.set_option(rs.option.visual_preset, 3)
```

**Does calling `pipeline.start(config)` invalidate existing device instances??**

I do not understand what the 'device' API contract is, so I am hoping to just learn more about how to use 'device' and how I can reliably understand if the 'device' instance is still valid.

Thank you.

Chandler, 2020-11-08:
Ah, I found something interesting here
"Devices of profiles, which are not returned by pipeline start() or get_active_profile(), are not guaranteed to be used by the pipeline." 
https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.pipeline_profile.html?highlight=get_device#pyrealsense2.pipeline_profile.get_device

This suggests that I do need to get my device only from pipeline.start()

MartyG-RealSense, 2020-11-09:
Hi @Chandler   If the problem is with the visual preset sometimes failing to set on the first attempt, perhaps you coud put in a back-up check that is structured something like the logic below (which is not proper Python code!) to set the preset as 3 if it currently does not equal (!=) a value of 3:

IF 
depth_sensor.get_option_range(rs.option.visual_preset) != 3

THEN
depth_sensor.set_option(rs.option.visual_preset, 3)

The link below contains a Python example of checking the visual preset:

https://github.com/IntelRealSense/librealsense/issues/2577

closed 2020-11-18 by Chandler