realsenseai/librealsense #7745: L515 pyrealsense2 SIGSEGV

quizzy5889, 2020-11-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  L515 | 
| Firmware Version                   | 01.05.01.03 | 
| Operating System & Version | MacOS Catalina 10.15.6  | 
| Kernel Version (Linux Only)    |  | 
| Platform                                 | PC |
| SDK Version                            | 2.39.0 | 
| Language                            | Python 3.8.6 | 
| Segment			|  Robot | 

### Issue Description
Hi all, I would like to develop a microservice which is responsible of acquiring data from my brand new L515. I'm working on macOS machine and therefore, I followed the instructions to build both librealsense and pyrealsense2. I also have here an older D435 depth camera that I used for an old project.

Now when running the "hello world" example, if the D435 is connected, then everything works as expected.
If the L515 I got a segmentation fault.

Any idea?

Process:               Python [10624]
Path:                  /Library/Frameworks/Python.framework/Versions/3.8/Resources/Python.app/Contents/MacOS/Python
Identifier:            Python
Version:               3.8.6 (3.8.6)
Code Type:             X86-64 (Native)
Parent Process:        pycharm [8346]
Responsible:           pycharm [8346]
User ID:               501

Date/Time:             2020-11-09 09:29:31.943 +0100
OS Version:            Mac OS X 10.15.6 (19G2021)
Report Version:        12
Bridge OS Version:     4.6 (17P6610)
Anonymous UUID:        2A1D4F21-FABB-DC47-D118-B7E5A52EBAB9

Sleep/Wake UUID:       57676CA2-FE00-42DF-ADB7-53DE2FAE3D8A

Time Awake Since Boot: 34000 seconds
Time Since Wake:       2300 seconds

System Integrity Protection: enabled

Crashed Thread:        13

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [10624]

VM Regions Near 0:
--> 
    __TEXT                 000000010d297000-000000010d298000 [    4K] r-x/rwx SM=COW  /Library/Frameworks/Python.framework/Versions/3.8/Resources/Python.app/Contents/MacOS/Python

----

I'll skip here till thread 13

----

Thread 13 Crashed:
0   librealsense2.2.39.0.dylib    	0x000000010dbf7bd2 std::__1::__function::__func<void dispatcher::invoke_and_wait<librealsense::platform::rs_hid_device::start_capture(std::__1::function<void (librealsense::platform::sensor_data const&)>)::$_2>(librealsense::platform::rs_hid_device::start_capture(std::__1::function<void (librealsense::platform::sensor_data const&)>)::$_2, std::__1::function<bool ()>, bool)::'lambda'(dispatcher::cancellable_timer), std::__1::allocator<void dispatcher::invoke_and_wait<librealsense::platform::rs_hid_device::start_capture(std::__1::function<void (librealsense::platform::sensor_data const&)>)::$_2>(librealsense::platform::rs_hid_device::start_capture(std::__1::function<void (librealsense::platform::sensor_data const&)>)::$_2, std::__1::function<bool ()>, bool)::'lambda'(dispatcher::cancellable_timer)>, void (dispatcher::cancellable_timer)>::operator()(dispatcher::cancellable_timer&&) + 1186 (functional:1720)
1   librealsense2.2.39.0.dylib    	0x000000010d9ecb13 dispatcher::dispatcher(unsigned int)::'lambda'()::operator()() const + 179 (concurrency.h:237)
2   librealsense2.2.39.0.dylib    	0x000000010d9ec9ed void* std::__1::__thread_proxy<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct> >, dispatcher::dispatcher(unsigned int)::'lambda'()> >(void*) + 45 (thread:286)
3   libsystem_pthread.dylib       	0x00007fff6ad17109 _pthread_start + 148
4   libsystem_pthread.dylib       	0x00007fff6ad12b8b thread_start + 15

ababilinski, 2020-11-13:
This native crash also happens in Unity

RealSenseSupport, 2021-01-12:
Hi @quizzy5889 

Anything else needed on this thread?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-19 by RealSenseSupport