realsenseai/librealsense #7748: [Unity][Android] Texture format r16 unsupported

TheBricktop, 2020-11-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Android pie  |                                  | 
| Platform                                 | /Raspberry Pi lineageos|
| SDK Version                            |2                         | 
| Language                            |  C#/unity                           | 
| Segment			|ar                          | 

### Issue Description
Trying the depth texture sample on android but texture shows null, Adb logcat tells me that texture format r16 is not supported on android. Is there a way to straightforward display depth data on android?

MartyG-RealSense, 2020-11-09:
Hi @TheBricktop   My understanding is that the best approach is to target Android from Unity using Java, and make sure to check USB permissions.

https://github.com/IntelRealSense/librealsense/issues/5516

There is a fully documented project created by a Realense community member that uses the Android-based Oculus Quest VR headset with Unity and D435i.

https://github.com/GeorgeAdamon/quest-realsense

For background information, the development process that led to that successful project is detailed in the discussion in the link below:

https://github.com/IntelRealSense/librealsense/issues/4155

TheBricktop, 2020-11-09:
well ive used coloriser and it works, just not the z16 texture format

dorodnic, 2020-11-10:
Hi @TheBricktop 
Single channel 16-bit format is not supported by all versions of underlying OpenGL ES drivers. Hence, putting raw depth data into Unity texture may not work on all devices. You can check what texture formats are supported using [SystemInfo.SupportsTextureFormat.html](https://docs.unity3d.com/ScriptReference/SystemInfo.SupportsTextureFormat.html)

closed 2020-11-26 by MartyG-RealSense