realsenseai/librealsense #7753: D435 D435i D415 all 3 cameras error on viewer 2.36 when connected to usb 3.2 gen2

johnjaiharjose, 2020-11-10:
Hr class error as soon as i connect any realsense camera to rear 10gbps usb3.2 gen2 or 5gbps usb3.1 gen1 ports.. I connect them direct with bundled cable.

I can see cameras are detected properly in device manager.

This is a new pc.

this issue doesnt happen on front usb3.0 but it shows up as usb2.1..

Old pc also doesnt have this problem (usb2.0 ports though.)

pc info:
Motherboard msi  z490-A-Pro
w10 pro
i7 10700k.

MartyG-RealSense, 2020-11-10:
Hi @johnjaiharjose   It should not be necessary to install the UWP driver in order to use RealSense 400 Series cameras with Windows.  This driver provides a means of interfacing RealSense with products that support Microsoft's Universal Windows Platform (UWP), such as the Hololens headset.  So most RealSense users will not need to have this driver installed.

There are also reported issues with errors when using the latest UWP driver version at the time of writing this **(6.11.160.21**).  So if you find that you need the UWP driver, I recommend using the older **6.1.160.14** version in the link below.

https://downloadcenter.intel.com/download/28723/Intel-RealSense-D400-Series-Universal-Windows-Platform-UWP-Driver-for-Windows-10

You should not need the UWP driver in order to successfully use the camera in USB ports though.  Could you remove the currently installed UWP driver please, do not install the older driver yet and let me know how that affects the computer.

johnjaiharjose, 2020-11-10:
> Hi @johnjaiharjose   It should not be necessary to install the UWP driver in order to use RealSense 400 Series cameras with Windows.  This driver provides a means of interfacing RealSense with products that support Microsoft's Universal Windows Platform (UWP), such as the Hololens headset.  So most RealSense users will not need to have this driver installed.
> 
> There are also reported issues with errors when using the latest UWP driver version at the time of writing this **(6.11.160.21**).  So if you find that you need the UWP driver, I recommend using the older **6.1.160.14** version in the link below.
> 
> https://downloadcenter.intel.com/download/28723/Intel-RealSense-D400-Series-Universal-Windows-Platform-UWP-Driver-for-Windows-10
> 
> You should not need the UWP driver in order to successfully use the camera in USB ports though.  Could you remove the currently installed UWP driver please, do not install the older driver yet and let me know how that affects the computer.
> 
> Also, did you install the RealSense SDK on Windows using the installer file from the SDK **Releases** page please, or by a different method?
> 
> https://github.com/IntelRealSense/librealsense/releases

I installed 2.36 via the installer file.

MartyG-RealSense, 2020-11-10:
I am not familiar with 'hr class' errors.  Do you mean **HResult** errors, please?

It would be very helpful if you could include screen images of the errors in a comment.

johnjaiharjose, 2020-11-11:
Sure i will upload them shortly. Give me 2-3 hours.

MartyG-RealSense, 2020-11-11:
Thanks very much @johnjaiharjose    :)

johnjaiharjose, 2020-11-11:
@MartyG-RealSense 

This is the actual error while connecting on any of the rear usb ports.. 
[image]

> 12:56:28 [Error] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\types.h:308 - MFCreateDeviceSource(_device_attrs, &_source) returned: HResult 0x80040154: "Class not registered"

This is how it works on the front case usb ports of the same computer. Only 15fps (depth + rgb) and 30 fps (depth + rgb but laggy depth). Shows Usb 2.1, connected to usb 3.0 port though.

[image]

firmware and other info
[image]

MartyG-RealSense, 2020-11-11:
The drivers look correct in the Device Manager.  There should be two drivers for the camera, Depth and RGB.  You have both present.

The **backend in rs2_create_device** error can occur if the Windows privacy settings have camera access set to Off.  Can you check if the ability of the camera to be accessed by applications is set to **On** in the privacy settings using the instructions in the link below please?

https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

johnjaiharjose, 2020-11-11:
> I am not familiar with 'hr class' errors.  Do you mean **HResult** errors, please?
> 
> It would be very helpful if you could include screen images of the errors 

yes i made sure camera app access is enabled already in w10 privacy settings.

MartyG-RealSense, 2020-11-11:
Thank you.  On the ports that do detect the camera as USB 2.1, does resetting the camera by selecting **Hardware Reset** from the Viewer's **More** menu cause them to be detected as USB 3 after reset?

johnjaiharjose, 2020-11-11:
> Thank you.  On the ports that do detect the camera as USB 2.1, does resetting the camera by selecting **Hardware Reset** from the Viewer's **More** menu cause them to be detected as USB 3 after reset?

Just tried this, no it shows same as usb 2.1 after hardware reset.

MartyG-RealSense, 2020-11-11:
In version 2.36 a built-in OpenVINO face detection system was introduced in the Viewer that is installed by the SDK installer that some computers can have problems with.  There is a standalone version of the Viewer that does not have the OpenVINO function and do may work on machines affected by this conflict.

You can download this 32 mb **Intel.RealSense.Viewer.exe** file from the same Assets file list that the SDK installer is in.  The downloaded Viewer .exe file should be launched from the location that it is downloaded to, and not with the desktop shortcut icon.

johnjaiharjose, 2020-11-11:
> In version 2.36 a built-in OpenVINO face detection system was introduced in the Viewer that is installed by the SDK installer that some computers can have problems with.  There is a standalone version of the Viewer that does not have the OpenVINO function and so may work on machines affected by this conflict.
> 
> You can download this 32 mb **Intel.RealSense.Viewer.exe** file from the same Assets file list that the SDK installer is in.  The downloaded Viewer .exe file should be launched from the location that it is downloaded to, and not with the desktop shortcut icon.

yep, this might be the problem. Because on my old computer i did not install full sdk components + just had the viewer downloaded seperately. I will try this one and update you on saturday.

MartyG-RealSense, 2020-11-11:
Thanks very much @johnjaiharjose - I look forward to hearing how it works out for you.  :)

johnjaiharjose, 2020-11-14:
@MartyG-RealSense 

Still not working, i have tried individual intelrealsenseviewer.exe versions of 2.39,2.36,2.29..... its the same error as before. Plus, i noticed some new error as soon as i launch the viewer it spams this too.
[image]

johnjaiharjose, 2020-11-14:
Its fixed!!!!!!..
I uninstalled the device manager drivers (right click uninstall on each of rgb and depth modules) but did not restart, then i launched the standalone viewer 2.36, reconnected the camera twice and voila!! it shows 3.2

Thank you @MartyG-RealSense 

[image]

MartyG-RealSense, 2020-11-14:
Awesome news @johnjaiharjose  - great to hear that you have succeeded.  Thanks for the update!

MartyG-RealSense, 2020-11-21:
Case closed due to a successful outcome and no further comments received.

closed 2020-11-21 by MartyG-RealSense