realsenseai/librealsense #7755: Reading just the RGB camera data from L515

ashwinj92, 2020-11-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 18) | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | NVIDIA Jetson |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I'm currently trying to read just the RGB camera and not the LIDAR data from L515 into Jetson Nano.  My question is can I do this with the existing code samples.

Any pointers or references with the existing documentation would be great.

RealSenseSupport, 2020-11-13:
Hi,

There isn't an example that does this specifically, but it's fairly straightforward to request different frames. If you look at this example:

https://github.com/IntelRealSense/librealsense/tree/master/examples/hello-realsense

simply replacing the calls/data for depth_frame get_depth_frame() with video_frame get_color_frame() should accomplish that.

Thanks

ashwinj92, 2020-11-15:
Thanks for the reply will try the mentioned approach.

closed 2020-12-02 by RealSenseSupport