realsenseai/librealsense #7756: we cannot transmit depth information and RGB information through pipe  at the same time

panxin0, 2020-11-11:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { l515} | 
| Firmware Version                   | 01.05.01.03 | 
| Operating System & Version |   Ubuntu 18.04 LTS  | 
| Kernel Version (Linux Only)    |   4.14.13  | 
| Platform                                 | PC  |
| SDK Version                            |  2.25 | 
| Language                            |  c++ opencv    | 
| Segment			|       | 

### Issue Description
<Describe your issue / question / feature request / etc..>

panxin0, 2020-11-11:
we cannot transmit depth information and RGB information through pipe at the same time,This will cause ghosting in the RGB window, similar to stitching together several images.However, it is fine to transmit RGB images separately. I want to know if there is a problem with the code, thank you！

RealSenseSupport, 2020-12-01:
Hi,

What do you mean by "at the same time" or "seperately"? Can you please elaborate or show a code sample of what you are trying to do?

Thanks

closed 2020-12-21 by RealSenseSupport