realsenseai/librealsense #7762: Intel ReakSense on Jetson Nano with Matlab

bladecell, 2020-11-12:
Hey, I wanted to ask, if it is possible to use Matlab GPU coder to generate CUDA code and run it on Jetson Nano with it using Intel Realsene camera. Thanks

MartyG-RealSense, 2020-11-12:
Hi @bladecell   If you build the RealSense SDK (librealsense) software on Jetson Nano from **packages** then CUDA support is enabled by default.  

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

If the RealSense SDK is built from source code using **CMake** then you can enable CUDA support by including the customization flag **-DBUILD_WITH_CUDA:bool=true** in the CMake build instruction.  The link below has an installation guide that builds librealsense and the Python compatibility wrapper and has been used successfully by Jetson Nano owners.

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

So if you were planning to use MATLAB GPU Coder just for the purposes of adding CUDA support to a RealSense application on Nano then it may be unnecessary, as the RealSense SDK already provides CUDA support on Jetson.

bladecell, 2020-11-12:
I want to write code that uses RealSense camera in Matlab, then use Matlab GPU Coder to run that code on Jetson.

MartyG-RealSense, 2020-11-12:
Okay, thank you very much for the clarification.   So would it be correct to say that you will not be using the RealSense SDK software at all, and your hope is to use MATLAB with a RealSense camera?  And so you want to know if the RealSense camera will work with your Nano and with MATLAB automatically if you just plug it into the USB port?

bladecell, 2020-11-12:
Basically. Realsense camera is connected to Nano. The Nano is connected to Matlab via ethernet (SSH I guess). I can access the RGB camera of D435 with no problem and I want to know, whether I can acces the depth camera. If I it were possible, I would then write code in Matlab and compile it in GPU coder and run that code on Jetson. I am not sure where I need the SDK, so I installed it on both devices (PC and Nano). I have succesfully connected the depth camera to Matlab, when the camera was plugged in the Windows PC I ran Matlab on.

MartyG-RealSense, 2020-11-12:
The RealSense MATLAB wrapper requires Windows 10, so this may not be an option for Jetsons which typically run Ubuntu.

There was a time before the existence of the MATLAB wrapper where RealSense data could interact with MATLAB but this was mainly limited to reading camera data from pre-recorded **rosbag** files rather than live data.  An example of this is a project called **bag2matlab**:

https://github.com/unl-nimbus-lab/bag2matlab

Another example from pre-wrapper days was code to read a **.raw** format file in MATLAB:

https://community.intel.com/t5/Items-with-no-label/Read-Raw-Depth-Image-with-MATLAB-Record-Stream/m-p/552442/highlight/true#M8792

Another way that RealSense users have read RealSense data into MATLAB in the past is to use the **Robotics System Toolbox**:

https://brisbaneroboticsclub.id.au/ros2-matlab-simulink-turtlebot/

https://github.com/IntelRealSense/librealsense/issues/1290#issuecomment-404903998

I am not sure how much of the above information will help you but hopefully it will spark some new ideas about how to proceed.

closed 2020-11-26 by MartyG-RealSense