realsenseai/librealsense #7766: D455 can not connected on the Jetson Xaiver

FrankCreen, 2020-11-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D455 } | 
| Firmware Version                   |  05.12.08.200  | 
| Operating System & Version |   Jetson  | 
| Kernel Version (Linux Only)    |  (e.g. 4.4.2)                                         | 
| Platform                                 |  NVIDIA Jetson / etc..  |
| SDK Version                            |         2.35.2           | 
| Language                            |  {C/opencv}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
  I followed the official document [https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md](url)

installed SDK on the jetson. And when I repluged D455 to the xaiver open the realsense-viewer,it print error message in the console window and exited.
```
realsense-viewer: /home/nvidia/Apps/librealsense-2.35.31/common/model-views.cpp:5105: rs2::device_model::draw_preset_panel(float, rs2::ux_window&, std::__cxx11::string&, rs2::viewer_model&, bool, bool, rs2::device_model::json_loading_func)::<lambda(rs2::option_model&, std::__cxx11::string&, rs2::notifications_model&)>: Assertion `opt_model.opt == RS2_OPTION_VISUAL_PRESET' failed.
Aborted (core dumped)
```

 I use the dmesg command and get these information:

```
[ 5340.965397] usb 2-4: new SuperSpeed USB device number 5 using tegra-xusb
[ 5340.985865] usb 2-4: New USB device found, idVendor=8086, idProduct=0b5c
[ 5340.985875] usb 2-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5340.985882] usb 2-4: Product: Intel(R) RealSense(TM) Depth Camera 455 
[ 5340.985887] usb 2-4: Manufacturer: Intel(R) RealSense(TM) Depth Camera 455 
[ 5340.985890] usb 2-4: SerialNumber: 025523052260
[ 5340.989029] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 5340.989041] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[ 5340.989230] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 455  (8086:0b5c)
[ 5340.994524] uvcvideo: Unable to create debugfs 2-5 directory.
[ 5340.994668] uvcvideo 2-4:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[ 5340.994848] uvcvideo 2-4:1.0: Entity type for entity Processing 2 was not initialized!
[ 5340.994997] uvcvideo 2-4:1.0: Entity type for entity Camera 1 was not initialized!
[ 5340.995350] input: Intel(R) RealSense(TM) Depth Ca as /devices/3610000.xhci/usb2/2-4/2-4:1.0/input/input14
[ 5340.996001] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[ 5340.996016] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 455  (8086:0b5c)
[ 5340.998497] uvcvideo: Unable to create debugfs 2-5 directory.
[ 5340.998615] uvcvideo 2-4:1.3: Entity type for entity Processing 7 was not initialized!
[ 5340.998762] uvcvideo 2-4:1.3: Entity type for entity Extension 8 was not initialized!
[ 5340.998907] uvcvideo 2-4:1.3: Entity type for entity Camera 6 was not initialized!
[ 5342.731387] mt7601u 1-1.1:1.0: Error: TSSI upper saturation
[ 5346.829794] mt7601u 1-1.1:1.0: Error: TSSI upper saturation
[ 5348.084967] usb 2-4: usb_suspend_both: status 0
[ 5348.085087] usb usb2: usb_suspend_both: status 0
```

I have read all the issues with the label "Jetson",and found many related methods,but no one tackle my problem. Waiting your help @MartyG-RealSense .  Thanks!

MartyG-RealSense, 2020-11-13:
Hi @FrankCreen   The link below has an installation guide that Nano users have had success with, and the guide's author stated that it also worked with their Xavier NX too:

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

FrankCreen, 2020-11-13:
> Hi @FrankCreen The link below has an installation guide that Nano users have had success with, and the guide's author stated that it also worked with their Xavier NX too. Have you tried this guide already, please?
> 
> [#6964 (comment)](https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049)

Thank you! 😀 I get the realsense-viewer work normally by the guide you referenced. I can get the video frame by realsense-viewer and the C++/Python interface. But one bug still exists, I run the same c++ code to capture the video frames (90FPS at a Intel Core i7 5500U with 2.4GHz)as yesterday only got 54FPS today. Why and what I can do? Thanks @MartyG-RealSense

MartyG-RealSense, 2020-11-13:
Great news that you had success with the Viewer @FrankCreen 

If you are using RGB at 90 FPS, you may find that it is reduced if RGB Auto-Exposure is **disabled** and **Auto-Exposure Priority** is **enabled**.

If RGB Auto-Exposure is **enabled** and Auto-Exposure Priority is **disabled** then this should force FPS to stay at around 90 constantly.

There is a D455 case about this from this week in the link below.

https://github.com/IntelRealSense/librealsense/issues/7749#issuecomment-725979247

FrankCreen, 2020-11-13:
> Great news that you had success with the Viewer @FrankCreen
> 
> I thought that you had solved the < 90 FPS problem earlier this week.
> 
> [#7749 (comment)](https://github.com/IntelRealSense/librealsense/issues/7749#issuecomment-726170529)

haha,the same code

MartyG-RealSense, 2020-11-13:
Okay, please update this case when you have some test results from your Xavier.  Good luck!

FrankCreen, 2020-11-16:
> Okay, please update this case when you have some test results from your Xavier. Good luck!

Same results——54FPS🤯 And the same code. I don't know what can I do next😫

MartyG-RealSense, 2020-11-16:
Regarding the **Invalid value in rs2_get_option** error: are you using a laptop with a built-in RGB webcam, or do you have a non-RealSense USB webcam attached to your computer, please?  I ask this because today I was running tests on another case and found that the **Invalid value in rs2_get_option** pop-up error box could occur if there was a conflict with a non-RealSense RGB webcam detected by the computer.  

It appeared when launching the Viewer even if there was no D455 plugged in.  In my case it was a built-in laptop webcam, and the error no longer appeared once I disabled the webcam in the computer's system settings (I was using Windows).

In regard to your FPS problem: could you try reducing the **Exposure** slider to a minimum of '1' in the Viewer please?  This will cause Auto-Exposure to disable, so re-enable Auto-Exposure again and then begin streaming and then see how FPS is affected.  I suggest this because manual exposure can drag down FPS speed.  The manual exposure values should not be affecting streaming if Auto-Exposure is enabled, but it is worth trying this to see if there is a problem that is causing the manual exposure to interfere.

FrankCreen, 2020-11-17:
> Regarding the **Invalid value in rs2_get_option** error: are you using a laptop with a built-in RGB webcam, or do you have a non-RealSense USB webcam attached to your computer, please? I ask this [...]

Nothing changed... I will test the 3.9 SDK next step and give the result.

FrankCreen, 2020-11-19:
@MartyG-RealSense  With tremendous coding test,I got a good result.
90FPS with color stream only，75FPS with color and depth stream simultaneously.
Thank you!

MartyG-RealSense, 2020-11-19:
Thnks very much for the report of your success with achieving 90 FPS color.   Do you need any further assistance please?  Thanks!

FrankCreen, 2020-11-19:
> Thnks very much for the report of your success with achieving 90 FPS color. Do you need any further assistance please? Thanks!

Not yet.This issue can br closed. Thank you!

MartyG-RealSense, 2020-11-19:
You are very welcome.  Thanks very much!

closed 2020-11-19 by MartyG-RealSense