realsenseai/librealsense #7767: Colorizing image relative from ground for machine learning

dream-guide, 2020-11-13:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |  Ubuntu 18.04.5 LTS | 
| Platform                                 | NVIDIA Jetson Nano |
| SDK Version                            |  2.0                          | 
| Language                            |  Python 3.6.9; Opencv 3.2.0    | 
| Segment			|  others                          | 

### Issue Description
I am trying to manipulate the depth image so that the colorized image is relative to the ground.
After that i want to train a deep neural network on these images.

I know the distance to the ground but the camera height can vary from 3m to 10m.
In the final image i want the ground to be always be a color, then up to 0,7m always orange and from there to 2m always a red gradient. 
Just a gradient starting at the ground up to 2m would be ok too.

Here an example:
[image]

I tried the colorizer but if i set the min and max distance nothing changes.
```
 colorizer = rs.colorizer()
 colorizer.set_option(rs.option.min_distance, value_min)
 colorizer.set_option(rs.option.max_distance, value_max)
 depth_colormap = np.asanyarray(colorizer.colorize(depth_frame).get_data())
```

I also tried with opencv and manipulating alpha and beta but have not found a way to translate that to real world measurements.
```
depth_colormap = cv.applyColorMap(cv.convertScaleAbs(depth_image, alpha=value_alpha, beta=value_beta), cv.COLORMAP_JET)
```

Does anyone have an idea that could help?

MartyG-RealSense, 2020-11-13:
Hi @dream-guide   You can have control over the color grading of depth with the colorizer by setting a **color scheme** and a **visual preset** for it.

The color schemes are selected with a number code (Jet is '0').  The full list of numbers is in the link below:

https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_processing.hpp#L717 

Below is a link to a discussion that contains a script in the Python language that includes setting the colorizer color scheme with **rs.option.color_scheme**

https://github.com/IntelRealSense/librealsense/issues/3363

For setting the colorizer's visual preset, the link below to a Python discussion on the subject will hopefully be helpful:

https://github.com/IntelRealSense/librealsense/issues/6874#issuecomment-661995064

dream-guide, 2020-11-13:
Hi @MartyG-RealSense Thank you that helped me with colorizer not using the min_distance and max_distance values.
I did not know visual_preset works for the depth sensor and colorizer.

Should i start a new issue for the question regarding the custom color scheme? 
I found out were they are defined in colorizer but i'm not sure how to define my own in python: 
https://github.com/IntelRealSense/librealsense/blob/master/src/proc/colorizer.cpp

This is the code that worked. Noteworthy is that the visual preset needs to be set before min and max.
```
colorizer = rs.colorizer()
colorizer.set_option(rs.option.visual_preset, 1) # 0=Dynamic, 1=Fixed, 2=Near, 3=Far
colorizer.set_option(rs.option.min_distance, value_min)
colorizer.set_option(rs.option.max_distance, value_max)

depth_colormap = np.asanyarray(colorizer.colorize(depth_frame).get_data())
```

MartyG-RealSense, 2020-11-13:
It is fine to discuss custom color schemes in this question, since it is a related subject.

I have not heard of a case where somebody has set their own custom colorization scale that is different from the SDK's, though someone did have the idea of exploring whether the source code of librealsense could be modified to do so.

https://github.com/cansik/realsense-processing/issues/24

dream-guide, 2020-11-16:
@MartyG-RealSense yes like in my example image the floor is solid green, objects smaller than 0.7m are orange and then it is a gradient.

I could go over the gradient image with numpy but if i use the colorizer anyways i thought i could do it in one go.
As i want to use this on runtime less operations are desirable.

MartyG-RealSense, 2020-11-17:
Apologies for the delay in responding further.  I carefully researched your question about a custom colorization scale but was unable to find an in-built way to do it in the RealSense SDK, unfortunately.

closed 2020-12-03 by MartyG-RealSense