realsenseai/librealsense #7768: d455 Auto exposure causes the image to be too dark

zhhz9706, 2020-11-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.08.200 | 
| Operating System & Version |  Linux (Ubuntu 18) | 
| Kernel Version (Linux Only)    |  (5.4.0-52)                                         | 
| Platform                                 | PC|
| SDK Version                            |  2.39     | 
| Language                            |  c++  | 
| Segment			|  robot                    | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hello, the camera I am using is RS D455. When I use the stereo camera, when the auto exposure is turned on, the image brightness of the stereo camera is not normal, it will be lower than the image brightness with the auto exposure turned off. Obviously, the exposure time is insufficient . In contrast, the exposure of an RGB camera is normal. 
Looking forward to your reply！

Enable automatic exposure
[image]
[image]
disable automatic exposure
[image]

RealSenseSupport, 2020-11-13:
@zhhz9706 The manual exposure value does not always follow the auto exposure value. If you turn off the auto exposure, you need to adjust the manual exposure value to get the best image.

dorodnic, 2020-11-15:
You may find this useful - https://dev.intelrealsense.com/docs/high-dynamic-range-with-stereoscopic-depth-cameras#section-2-4-manual-vs-auto-exposure

sam598, 2020-11-15:
> The manual exposure value does not always follow the auto exposure value.

While the value doesn't match, in my own anecdotal experience auto exposure on the D455 is always aggressively under-exposed, which leads to a big loss in depth map quality.

The HDR mode does help, but when capturing moving subjects where temporal coherence is required, manual exposure seems to be the only valid option.

agrunnet, 2020-11-15:
If you want to change the set point for the autoexposure you can do that too.

zhhz9706, 2020-11-16:
> @zhhz9706 The manual exposure value does not always follow the auto exposure value. If you turn off the auto exposure, you need to adjust the manual exposure value to get the best

zhhz9706, 2020-11-16:
> @zhhz9706 The manual exposure value does not always follow the auto exposure value. If you turn off the auto exposure, you need to adjust the manual exposure value to get the best image.

Hello, thanks for your reply.
What I want to say is that when auto exposure is enabled, the camera's exposure is not enough, causing the image to look dark. For disable auto exposure, you can adjust to obtain a relatively normal image brightness.
Thanks.

closed 2020-11-16 by zhhz9706

RealSenseSupport, 2020-11-16:
We intended to make the IR (left and right) images a little bit darker than normal RGB image. A little darker IR images will generate a better depth. As @agrunnet suggested, you can try a higher set point to compare the depth maps.

sam598, 2020-11-16:
@RealSenseSupport It's subjective but from my own experience the auto exposed IR images are significantly underexposed, and the darker images produce worse depth. 

> If you want to change the set point for the autoexposure you can do that too.

Does this value offset the auto exposure value by stops? Or does it change how the algorithm behaves? In the latest version of RealSenseViewer it does not seem like this value can be edited.