realsenseai/librealsense #7770: Export PLY pointcloud without mesh coloured by RGB camera

eliotts, 2020-11-13:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | 05.12.07.100) | 
| Operating System & Version |   {Win (10) /3  | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.36.0 }                          | 
| Language                            |  {python }                        | 
| Segment			|  {Robot}                          | 

### Issue Description
Hi @RealSenseSupport 
I am trying to export a pointcloud with the RGB colors from the camera to a PLY file. 
Unfortunately for now I am only able to get the pointcloud without any colors.
I am working upon example code provided in the librealsense repository but I am not able to find the correct way to align and the colors to my pointcloud. 
What would be the proper way to have the RGB colors mapped onto the pointcloud?
Here is the code I am working on.

```
import numpy as np                        # fundamental package for scientific computing
import matplotlib.pyplot as plt           # 2D plotting library producing publication quality figures
import pyrealsense2 as rs                 # Intel RealSense cross-platform open-source API

import numpy as np

print("Environment Ready")

def export (mesh=False, pc=True):
    # Setup:
    pipe = rs.pipeline()
    cfg = rs.config()
    cfg.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
    # cfg.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 30)
    cfg.enable_stream(rs.stream.color, 640, 480, rs.format.rgb8, 30)
    profile = pipe.start(cfg)

    # Create an align object
    # rs.align allows us to perform alignment of depth frames to others frames
    # The "align_to" is the stream type to which we plan to align depth frames.
    align_to = rs.stream.color
    align = rs.align(align_to)

    # Skip 5 first frames to give the Auto-Exposure time to adjust
    for x in range(5):
        pipe.wait_for_frames()

    frames = []
    for x in range(15):
        frameset = pipe.wait_for_frames()
        aligned_frames = align.process(frameset)
        frames.append(aligned_frames.get_depth_frame())
        frames.append(aligned_frames.get_color_frame())

    # Cleanup:
    pipe.stop()
    print("Frames Captured")

    ##DECLARE FILTERS
    #Threshold filter
    threshold = rs.threshold_filter(min_dist = 0.15, max_dist = 2)
    #Decimation filter
    decimation = rs.decimation_filter()
    decimation.set_option(rs.option.filter_magnitude, 2)
    #Spatial filter
    spatial = rs.spatial_filter()
    spatial.set_option(rs.option.filter_magnitude, 1)#Filtering itterations from 1.000 to 5.000
    spatial.set_option(rs.option.filter_smooth_alpha, 1)
    spatial.set_option(rs.option.filter_smooth_delta, 3)# 8 is good
    spatial.set_option(rs.option.holes_fill, 2)
    #Temporal Filter
    temporal = rs.temporal_filter()
    temporal.set_option(rs.option.filter_smooth_alpha, 0.2)
    temporal.set_option(rs.option.filter_smooth_delta, 100)
    #Hole filling
    hole_filling = rs.hole_filling_filter()
    #Colorize
    colorizer = rs.colorizer()
    colorizer.set_option(rs.option.color_scheme, 2)
    rs.disparity_transform(True)
    rs.disparity_transform(True)

    #Apply filters
    for i in range(15):
        depth_frame = frames[i]
        depth_frame = threshold.process(depth_frame)
        depth_frame = temporal.process(depth_frame)
        depth_frame = decimation.process(depth_frame)
        depth_frame = spatial.process(depth_frame)
    
    # Create save_to_ply object
    ply = rs.save_to_ply("C://pc.ply")

    # Set options to the desired values
    # Generate a textual PLY with normals (mesh is already created by default)
    ply.set_option(rs.save_to_ply.option_ply_mesh, False)
    ply.set_option(rs.save_to_ply.option_ply_binary, False)
    ply.set_option(rs.save_to_ply.option_ply_normals, True)
    ply.set_option(rs.save_to_ply.option_ignore_color, False)

    print("Saving ply...")
    # Apply the processing block to the frameset which contains the depth frame and the texture
    ply.process(depth_frame)
    print("Done")

export()
```
>

RealSenseSupport, 2020-11-30:
@eliotts Could you please try export_to_ply instead? I can get pointcloud colored by RGB with the attached code snippet. Could you please have a try? Thanks!

[testalign.txt](https://github.com/IntelRealSense/librealsense/files/5614907/testalign.txt)

eliotts, 2020-11-30:
@RealSenseSupport ,
Thank you for your answer.
 This is actually the way we found to work around this export problem, but what we don't like about this export_to_ply method  is that :
1) we are not able to save Point Normals, and
2) that you are not able to export in ASCII which makes it then impossible to import to certain CAD softwares. 
It would be really great to have a way to use save_to_ply with all the same options as in the Intel RealsenseViewer.

RealSenseSupport, 2020-12-07:
@eliotts  Could you please try if use _frameset_ instead of **depth_frame** for ply.process as below is what you want? 

ply.process(frameset)

RealSenseSupport, 2020-12-15:
@eliotts Any update from your side? Looking forward to your reply. Thanks!

RealSenseSupport, 2020-12-22:
@eliotts Could you please update? Please note that the issue will be closed if we don't hear from you in another 7 days. Thanks!

RealSenseSupport, 2020-12-30:
@eliotts Solution provided. Sorry we don't hear from you for several weeks. Will close it at this point. Please create new one if you still have questions or issues. Thanks!

closed 2020-12-30 by RealSenseSupport

eliotts, 2021-01-06:
Hi @RealSenseSupport ,

Sorry for the long silence,

after trying your solution I am not able to apply the pos production filters anymore. Would you have a solution to apply the filters directly to the frameset ?

agenis, 2021-03-03:
@RealSenseSupport Hi, 
I'm interested in your example solution that you provided in the testaligned.txt file above. 
would you mind adapting this script for the L515 for which the options for RGB and depth resolutions are not matching (
`320*240 640*480 1024*768` for depth and 
`960*540 1280*720 1920*1080` for RGB) and the script doesn't work as such. Thanks