realsenseai/librealsense #7771: Threshold on Depth Image

sdkaplan, 2020-11-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version | Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                        | 
| Platform                                 | Nvidia Jetson Xavier  |
| SDK Version                            |  2.35.2                      | 
| Language                            |  C++     | 

I want to threshold the depth between two values without using realsense-ros. I currently have the code below. I am applying the filter to one depth map and not the other and then converting both images to opencv Mat so I can display using cv2::imshow(). The issue is that the threshold filter does not seem to be working. I am applying the filter and changing its parameters but nothing changes in the resulting image. Any suggestions? Am I using this filter wrong?

Code:
`int main(int argc, char **argv) try
{
    // SET UP REALSENSE
    rs2::colorizer color_map; // Declare depth colorizer for pretty visualization of depth
    rs2::pipeline pipe; // Declare RealSense pipeline, encapsulating the device and sensors
    // Start streaming with default recommended configuration. The default video configuration contains Depth and Color streams, Gyro and Accelerometer are enabled by default
    pipe.start();

    rs2::decimation_filter dec_filter;
    rs2::spatial_filter spat_filter;
    rs2::spatial_filter spat_filter_alpha;
    rs2::threshold_filter thres_filter(0.15f, 4.f);
    rs2::temporal_filter temp_filter;
     
    rs2::frameset data = pipe.wait_for_frames();    // Wait for next set of frames from the camera
    rs2::frame depth = data.get_depth_frame().apply_filter(color_map);
    // Query frame size (width and height)
    const int w = depth.as<rs2::video_frame>().get_width();
    const int h = depth.as<rs2::video_frame>().get_height();

    spat_filter_alpha.set_option(RS2_OPTION_FILTER_SMOOTH_ALPHA, 1);

    spat_filter.set_option(RS2_OPTION_HOLES_FILL, 2);

    while (waitKey(1) < 0){
        rs2::frameset data = pipe.wait_for_frames();// Wait for next set of frames
        rs2::frame depth = data.get_depth_frame().apply_filter(color_map); //rs depth frame
        rs2::frame depth1 = data.get_depth_frame().apply_filter(color_map); //rs depth frame
        depth = spat_filter_alpha.process(depth);
        depth = spat_filter.process(depth);    
        depth = thres_filter.process(depth);

        Mat frame_depth1(Size(w, h), CV_8UC3, (void*)depth1.get_data(), Mat::AUTO_STEP);
        Mat frame_depth(Size(w, h), CV_8UC3, (void*)depth.get_data(), Mat::AUTO_STEP);

	    // Save frame to video
        imshow("depth filtered", frame_depth);
        imshow("depth raw", frame_depth1);

        //cv_ptr->image = frame_depth;
        //cv_ptr->encoding = "bgr8";
        //image_pub_.publish(cv_ptr->toImageMsg());

        // Exit if ESC pressed.
        int k = waitKey(1);
        if(k == 27){
            break;
        }

    }

}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}
`

MartyG-RealSense, 2020-11-14:
Hi @sdkaplan    Does the threshold filter work if you configure the minimum and maximum with the C++ instructions below, please?

**thres_filter.set_option(RS2_OPTION_MIN_DISTANCE, 0.15);       
thres_filter.set_option( RS2_OPTION_MAX_DISTANCE, 4.0);**

dorodnic, 2020-11-15:
I would also recommend applying threshold filter before colour map

sdkaplan, 2020-11-16:
I applied both suggestions and it works now! Thanks so much!

closed 2020-11-16 by sdkaplan

MartyG-RealSense, 2020-11-16:
That's great to hear @sdkaplan - thanks for the update!