realsenseai/librealsense #7772: No device connected - Intel Realsense Tracking Camera T265, when trying to run rs-pose Linux, Ubuntu

juliahambright, 2020-11-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | | 
| Operating System & Version |   Ubuntu 18.04.4  | 
| Kernel Version (Linux Only)    |  5.3                                       | 
| Platform                                 | Raspberry Pi 3 b+ |
| SDK Version                            |  Intel® RealSense™ SDK 2.0 | 
| Language                            |  C++ | 
| Segment			|  Camera | 

### Issue Description
The first time I plug in my T265 camera and run rs-pose (an example program using the camera), it works and my camera is recognized, and when I run "lsusb" this is what is outputted:

Bus 001 Device 005: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 001 Device 006: ID 03e7:2150  
Bus 001 Device 004: ID 0424:7800 Standard Microsystems Corp. 
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

However, when I run rs-pose the second or sometimes third time, it no longer recognizes the camera, and I get
"RealSense error calling rs2_pipeline_start_with_config(pipe:0x7df848, config:0x7bb630):
    No device connected"

Then when I run "lsusb," the output is this:

Bus 001 Device 005: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 001 Device 007: ID 8087:0b37 Intel Corp. 
Bus 001 Device 004: ID 0424:7800 Standard Microsystems Corp. 
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Whenever I built from the source and ran the cmake command, I used this:

"cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true" 

from the suggestion of another post with this same issue.

dorodnic, 2020-11-15:
Hi @juliahambright 
Have you installed the relevant udev-rules:
https://github.com/IntelRealSense/librealsense/blob/master/scripts/libuvc_installation.sh#L42-L43

juliahambright, 2020-11-15:
Hi @dorodnic 
I believe so, I'm sorry I meant to attach the link I used to build from the source: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

I think the udev rules were installed in step 3.

I did get stuck on step 4 though, and I found a post that said to use this cmake instead as a work around for applying the patched kernel modules: "cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true" but I don't think that would have affected step 3.

Let me know what you think.

Thanks

juliahambright, 2021-02-09:
Update: I had to install this code: https://askubuntu.com/questions/645/how-do-you-reset-a-usb-device-from-the-command-line called usbreset, and I run it after every time I run the camera. It resets the usb and is able to recognize the camera.

closed 2021-02-09 by juliahambright