realsenseai/librealsense #7775: Unable to build pyrealsense2 for python3.6

prparsons, 2020-11-16:
D435i (05.12.07.100)
Ubuntu 18.04 (Jetson Nano)

Cloned the repo yesterday after realizing the apt package doesn't contain pyrealsense2. After building I'm unable to get pyrealsense2 working in python 3.6, I keep getting python 2 lib files

What I used to build (yes, that's the correct path to python3.6):
cmake ../ -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=/usr/bin/python3.6
make -j4
sudo make install

The only place it installs pyrealsense.so files is in /usr/local/lib/python2.7/pyrealsense2/
The pyrealsense.so files in build/wrappers/python/ only work for python 2. If I try import pyrealsense2 in python3.6, i get:
ImportError: /home/ryan/librealsense/build/wrappers/python/pyrealsense2.so: undefined symbol: _Py_ZeroStruct

MartyG-RealSense, 2020-11-16:
Hi @prparsons   There is an installation guide in the link below that has worked for Nano users.

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049

If that does not work for you, there is another Jetson guide that targets Python 3.6:

https://github.com/IntelRealSense/librealsense/issues/7722

prparsons, 2020-11-19:
Got a working version of the lib (it still isn't installing it in /usr/local/lib/python3.6 on its own for some reason, but that's not a big deal since it's in the wrappers/python/ directory).

Seems like the main issue was the version of cmake, the version installed by apt (3.10.2-1ubuntu2.18.04.1) won't work. When I updated to use 3.13.0 as indicated by your first link, that finally got it to build the python3 lib.

One other thing to mention, initially it couldn't find the cuda compiler, adding the following to bashrc fixed that:
export CUDA_HOME=/usr/local/cuda
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/cuda/lib64:/usr/local/cuda/extras/CUPTI/lib64
export PATH=$PATH:$CUDA_HOME/bin

I'll actually test it tonight, verify it works, but I'm assuming it will since python3 is successfully importing it now. Thanks for your help!

MartyG-RealSense, 2020-11-20:
Thanks very much for the update!  I look forward to your next report.

MartyG-RealSense, 2020-11-26:
Hi @prparsons   Do you have an update about this case that you can provide, please?   Thanks!

prparsons, 2020-11-26:
Oh yeah, everything worked out just fine, imports, runs everything it should (the desktop environment stopped working on the nano for some reason so I had to fix that before I verified everything).

MartyG-RealSense, 2020-11-26:
Great news - I will close this case then as you have achieved a successful outcome.   Thanks for the update!

closed 2020-11-26 by MartyG-RealSense