realsenseai/librealsense #7776: Jetson TX2 install realsense error, unable to display RGB image

GaoZiHong, 2020-11-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   |    05.12.06.00   | 
| Operating System & Version |   Ubuntu 16.04.7 LTS  | 
| Kernel Version (Linux Only)    | 4.4.38+     | 
| Platform                                 | NVIDIA Jetson TX2  |
| SDK Version                            |                 | 
| Language                            | C C++            | 
| Segment			|  Robot                       | 

### Issue Description
My system version is JetPack 3.2.1. I installed SDK and librealsense (v2.13.0) correctly according to the method provided by https://github.com/jetsonhacks/buildLibrealsense2TX, but realsense-viewer cannot detect the camera. After consulting the information, I found that the version required to use D435i is at least 2.16.0. I completely uninstalled the v2.13.0 installed before, and then installed librealsense (v2.39.0) correctly on the official website using the reusb method. I opened realsense- The viewer can detect the camera correctly. I run the depth camera and get the depth map correctly. At this time, I wanted to run the RGB camera, but found that the RGB camera could not be opened and started reporting errors. The error message is as follows:
08:08:04 [Info]-librealsense version: 2.39.0

08:08:04 [Warn] /home/dji/librealsense_build/librealsense-master/common/notifications.cpp:514-Multiple realsense udev-rules were found!:
1:/etc/udev/rules.d/99-realsense-libusb.rules
2: /lib/udev/rules.d/60-librealsense2-udev-rules.rules
Make sure to remove redundancies!
08:08:04 [Info] /home/dji/librealsense_build/librealsense-master/src/rs.cpp:2552-Framebuffer size changed to 1344 x 756
08:08:04 [Info] /home/dji/librealsense_build/librealsense-master/src/rs.cpp:2552-Scale Factor is now 1
08:08:05 [Info] /home/dji/librealsense_build/librealsense-master/src/context.cpp:365-Found 1 RealSense devices (mask ff)
08:08:05 [Info] /home/dji/librealsense_build/librealsense-master/common/notifications.cpp:541-Intel RealSense D435I (S/N 902512070115)
Current Version: 05.12.06.00
Recommended Version: 5.12.8.200
08:08:05 [Info]-Intel RealSense D435I was selected as a default device
08:12:17 [Info]-Starting streaming of Color...
08:12:17 [Info] /home/dji/librealsense_build/librealsense-master/src/sensor.cpp:1342-Request: RGB8 Color,
Resolved to: YUYV Color,
08:12:17 [Info] /home/dji/librealsense_build/librealsense-master/src/uvc/uvc-streamer.cpp:28-endpoint 84 read buffer size: 1844224
08:12:17 [Error] /home/dji/librealsense_build/librealsense-master/src/proc/synthetic-stream.cpp:48-Exception was thrown during user processing callback: � 
08:12:17 [Error] /home/dji/librealsense_build/librealsense-master/src/proc/synthetic-stream.cpp:48-Exception was thrown during user processing callback: � 
08:12:17 [Error] /home/dji/librealsense_build/librealsense-master/src/proc/synthetic-stream.cpp:48-Exception was thrown during user processing callback: � 
08:12:17 [Error] /home/dji/librealsense_build/librealsense-master/src/proc/synthetic-stream.cpp:48-Exception was thrown during user processing callback: � 
08:12:18 [Error] /home/dji/librealsense_build/librealsense-master/src/proc/synthetic-stream.cpp:48-Exception was thrown during user processing callback: � 
08:12:18 [Error] /home/dji/librealsense_build/librealsense-master/src/proc/synthetic-stream.cpp:48-Exception was thrown during user processing callback: � 
08:12:18 [Error] /home/dji/librealsense_build/librealsense-master/src/proc/synthetic-stream.cpp:48-Exception was thrown during user processing callback: � 
08:12:18 [Error] /home/dji/librealsense_build/librealsense-master/src/proc/synthetic-stream.cpp:48-Exception was thrown during user processing callback: � 
08:12:18 [Error] /home/dji/librealsense_build/librealsense-master/src/proc/synthetic-stream.cpp:48-Exception was thrown during user processing callback: � 
08:12:18 [Error] /home/dji/librealsense_build/librealsense-master/src/proc/synthetic-stream.cpp:48-Exception was thrown during user processing callback: � 
08:12:18 [Error] /home/dji/librealsense_build/librealsense-master/src/proc/synthetic-stream.cpp:48-Exception was thrown during user processing callback: � 
08:12:18 [Error] /home/dji/librealsense_build/librealsense-master/src/proc/synthetic-stream.cpp:48-Exception was thrown during user processing callback: � 
08:12:18 [Error] /home/dji/librealsense_build/librealsense-master/src/proc/synthetic-stream.cpp:48-Exception was thrown during user processing callback: � 
08:12:18 [Error] /home/dji/librealsense_build/librealsense-master/src/proc/synthetic-stream.cpp:48-Exception was thrown during user processing callback: � 
08:12:18 [Error] /home/dji/librealsense_build/librealsense-master/src/proc/synthetic-stream.cpp:48-Exception was thrown during user processing callback: � 
08:12:18 [Error] /home/dji/librealsense_build/librealsense-master/src/proc/synthetic-stream.cpp:48-Exception was thrown during user processing callback: � 
08:12:18 [Error] /home/dji/librealsense_build/librealsense-master/src/proc/synthetic-stream.cpp:48-Exception was thrown during user processing callback: � 
08:12:18 [Error] /home/dji/librealsense_build/librealsense-master/src/proc/synthetic-stream.cpp:48-Exception was thrown during user processing callback: � 
08:12:19 [Error] /home/dji/librealsense_build/librealsense-master/src/proc/synthetic-stream.cpp:48-Exception was thrown during user processing callback: � 
08:12:19 [Error] /home/dji/librealsense_build/librealsense-master/src/proc/synthetic-stream.cpp:48-Exception was thrown during user processing callback: � 
08:12:19 [Error] /home/dji/librealsense_build/librealsense-master/src/proc/synthetic-stream.cpp:48-Exception was thrown during user processing callback: � 
08:12:19 [Error] /home/dji/librealsense_build/librealsense-master/src/proc/synthetic-stream.cpp:48-Exception was throw

GaoZiHong, 2020-11-16:
The results of executing rs-enumerate-devices are as follows:
Device info:
    Name: Intel RealSense D435I
    Serial Number: 902512070115
    Firmware Version: 05.12.06.00
    Recommended Firmware Version: 05.12.08.200
    Physical Port: 2-2.3-12
    Debug Op Code: 15
    Advanced Mode: YES
    Product Id: 0B3A
    Camera Locked: YES
    Usb Type Descriptor: 3.2
    Product Line: D400
    Asic Serial Number: 915323051809
    Firmware Update Id: 915323051809

Stream Profiles supported by Stereo Module
 Supported modes:
    stream resolution fps format
    Infrared 1 1280x800 @ 30Hz Y8
    Infrared 1 1280x800 @ 25Hz Y16
    Infrared 1 1280x800 @ 15Hz Y16
    Infrared 1 1280x800 @ 15Hz Y8
    Infrared 1 1280x720 @ 30Hz Y8
    Infrared 1 1280x720 @ 15Hz Y8
    Infrared 1 1280x720 @ 6Hz Y8
    Infrared 1 848x480 @ 90Hz Y8
    Infrared 1 848x480 @ 60Hz Y8
    Infrared 1 848x480 @ 30Hz Y8
    Infrared 1 848x480 @ 15Hz Y8
    Infrared 1 848x480 @ 6Hz Y8
    Infrared 1 848x100 @ 300Hz Y8
    Infrared 1 848x100 @ 100Hz Y8
    Infrared 1 640x480 @ 90Hz Y8
    Infrared 1 640x480 @ 60Hz Y8
    Infrared 1 640x480 @ 30Hz Y8
    Infrared 1 640x480 @ 15Hz Y8
    Infrared 1 640x480 @ 6Hz Y8
    Infrared 1 640x400 @ 25Hz Y16
    Infrared 1 640x400 @ 15Hz Y16
    Infrared 1 640x360 @ 90Hz Y8
    Infrared 1 640x360 @ 60Hz Y8
    Infrared 1 640x360 @ 30Hz Y8
    Infrared 1 640x360 @ 15Hz Y8
    Infrared 1 640x360 @ 6Hz Y8
    Infrared 1 480x270 @ 90Hz Y8
    Infrared 1 480x270 @ 60Hz Y8
    Infrared 1 480x270 @ 30Hz Y8
    Infrared 1 480x270 @ 15Hz Y8
    Infrared 1 480x270 @ 6Hz Y8
    Infrared 1 424x240 @ 90Hz Y8
    Infrared 1 424x240 @ 60Hz Y8
    Infrared 1 424x240 @ 30Hz Y8
    Infrared 1 424x240 @ 15Hz Y8
    Infrared 1 424x240 @ 6Hz Y8
    Infrared 1 256x144 @ 300Hz Y8
    Infrared 1 256x144 @ 90Hz Y8
    Infrared 2 1280x800 @ 30Hz Y8
    Infrared 2 1280x800 @ 25Hz Y16
    Infrared 2 1280x800 @ 15Hz Y16
    Infrared 2 1280x800 @ 15Hz Y8
    Infrared 2 1280x720 @ 30Hz Y8
    Infrared 2 1280x720 @ 15Hz Y8
    Infrared 2 1280x720 @ 6Hz Y8
    Infrared 2 848x480 @ 90Hz Y8
    Infrared 2 848x480 @ 60Hz Y8
    Infrared 2 848x480 @ 30Hz Y8
    Infrared 2 848x480 @ 15Hz Y8
    Infrared 2 848x480 @ 6Hz Y8
    Infrared 2 848x100 @ 300Hz Y8
    Infrared 2 848x100 @ 100Hz Y8
    Infrared 2 640x480 @ 90Hz Y8
    Infrared 2 640x480 @ 60Hz Y8
    Infrared 2 640x480 @ 30Hz Y8
    Infrared 2 640x480 @ 15Hz Y8
    Infrared 2 640x480 @ 6Hz Y8
    Infrared 2 640x400 @ 25Hz Y16
    Infrared 2 640x400 @ 15Hz Y16
    Infrared 2 640x360 @ 90Hz Y8
    Infrared 2 640x360 @ 60Hz Y8
    Infrared 2 640x360 @ 30Hz Y8
    Infrared 2 640x360 @ 15Hz Y8
    Infrared 2 640x360 @ 6Hz Y8
    Infrared 2 480x270 @ 90Hz Y8
    Infrared 2 480x270 @ 60Hz Y8
    Infrared 2 480x270 @ 30Hz Y8
    Infrared 2 480x270 @ 15Hz Y8
    Infrared 2 480x270 @ 6Hz Y8
    Infrared 2 424x240 @ 90Hz Y8
    Infrared 2 424x240 @ 60Hz Y8
    Infrared 2 424x240 @ 30Hz Y8
    Infrared 2 424x240 @ 15Hz Y8
    Infrared 2 424x240 @ 6Hz Y8
    Infrared 2 256x144 @ 300Hz Y8
    Infrared 2 256x144 @ 90Hz Y8
    Depth 1280x720 @ 30Hz Z16
    Depth 1280x720 @ 15Hz Z16
    Depth 1280x720 @ 6Hz Z16
    Depth 848x480 @ 90Hz Z16
    Depth 848x480 @ 60Hz Z16
    Depth 848x480 @ 30Hz Z16
    Depth 848x480 @ 15Hz Z16
    Depth 848x480 @ 6Hz Z16
    Depth 848x100 @ 300Hz Z16
    Depth 848x100 @ 100Hz Z16
    Depth 640x480 @ 90Hz Z16
    Depth 640x480 @ 60Hz Z16
    Depth 640x480 @ 30Hz Z16
    Depth 640x480 @ 15Hz Z16
    Depth 640x480 @ 6Hz Z16
    Depth 640x360 @ 90Hz Z16
    Depth 640x360 @ 60Hz Z16
    Depth 640x360 @ 30Hz Z16
    Depth 640x360 @ 15Hz Z16
    Depth 640x360 @ 6Hz Z16
    Depth 480x270 @ 90Hz Z16
    Depth 480x270 @ 60Hz Z16
    Depth 480x270 @ 30Hz Z16
    Depth 480x270 @ 15Hz Z16
    Depth 480x270 @ 6Hz Z16
    Depth 424x240 @ 90Hz Z16
    Depth 424x240 @ 60Hz Z16
    Depth 424x240 @ 30Hz Z16
    Depth 424x240 @ 15Hz Z16
    Depth 424x240 @ 6Hz Z16
    Depth 256x144 @ 300Hz Z16
    Depth 256x144 @ 90Hz Z16

Stream Profiles supported by RGB Camera
 Supported modes:
    stream resolution fps format
    Color 1920x1080 @ 30Hz RGB8
    Color 1920x1080 @ 30Hz RAW16
    Color 1920x1080 @ 30Hz Y16
    Color 1920x1080 @ 30H

MartyG-RealSense, 2020-11-16:
Hi @GaoZiHong   The error **Multiple realsense udev-rules were found** may occur if you installed librealsense from packages and also installed it from source code (which the **RSUSB** method would do).

You can make the multiple udev error disappear by removing the udev rules located at **etc/udev/rules.d/99-realsense-libusb.rules**.  

You should not remove the set of rules at **/lib/udev/rules.d/60-librealsense2-udev-rules.rules**.

Further details can be found at the link below.

https://github.com/IntelRealSense/librealsense/issues/6153#issuecomment-605687547

GaoZiHong, 2020-11-16:
thank you. I know how to solve this problem. My question is why my RGB camera cannot be enabled after I follow the operation I described before. When I try to enable it, the above error appears, indicating that there is a problem with my user data callback .

MartyG-RealSense, 2020-11-16:
Do you activate the Motion Module at all before the RGB problem with **Exception was thrown during user processing callback** occurs, please?

GaoZiHong, 2020-11-17:
Thank you. I don’t know if I have activated the motion module. I just run realsen-viewer after installing it according to the installation tutorial, and then the above problem occurs. How should I check if I have activated the motion module to troubleshoot this problem?

发自我的 iPad

MartyG-RealSense, 2020-11-17:
If the Motion Module is activated then it has a blue (On) icon beside it instead of a black (Off) icon.  The Motion Module, like the other streams, is always Off (black) when the RealSense Viewer launches.

[image]

GaoZiHong, 2020-11-18:
Thank you, when I turn on the motion module, the depth image can be displayed normally, but the RGB image still cannot be displayed, and if the depth map and RGB image are opened at the same time, the depth map will also crash and eventually show that the camera loses connection. When opening the RGB image, the error message appears as described above

发自我的iPhone

MartyG-RealSense, 2020-11-18:
Thank you for the additional details.  Your experience with the RealSense Viewer is a known issue when activating the Motion Module.  There are programming workarounds in C++ and Python when writing your own application but there is not currently a solution for the Viewer.  So the best recommendation would be to only use the Motion Module on its own in the Viewer.  

If you activate color, depth, or both color and depth at the same time, then you should not have problems with the RGB if you leave the Motion Module unactivated.

GaoZiHong, 2020-11-18:
Thank you, but I don't quite understand what you said. Do you mean I can only use the motion module alone in realsense-viewer? But RGB image and depth image are necessary for me. And my problem is that after I activate the motion module, I still cannot get the RGB image, and the above error message appears, showing the user callback error. To avoid misunderstandings, I repeat my question. RGB cameras cannot be used, including all the following situations:
1. Only open the RGB image, do not activate the motion module, RGB cannot be used;
2. Activate the motion module and open the RGB image;
3. Activate the motion module and open the RGB image and the depth image at the same time, the depth image cannot be used at this time;
I didn't change any settings. After I installed librealsense, this happened. In all cases, the RGB camera could not be used, and the error message I got was the same. I want to know how to solve this problem, and what is the reason for this problem, why is there a user data callback error? But can the depth map be used normally? Is it the kernel of my system? Please help me to solve this problem, it is very important to me, thank you

发自我的iPhone

MartyG-RealSense, 2020-11-18:
I do not mean not to use depth and color.  I mean to use depth and color in one Viewer session, and then close and re-open the Viewer and use motion module on its own (do not use the motion module at the same time as depth or color).  Do not turn the motion module on when you are using depth or color.

You have not done anything wrong and your computer is not at fault.  This is a well known issue with the Viewer.

GaoZiHong, 2020-11-18:
Thank you for your quick reply. You mean I can’t activate the motion module and the RGB camera in the same viewer window? But I remember I did this when I used realsense on the x86 architecture. If your operation is correct, it means that my current system configuration allows me to use the RGB camera normally, right?

发自我的iPhone

MartyG-RealSense, 2020-11-18:
Yes, please avoid activating the motion module and RGB in the same window of the Viewer if doing so disrupts the RGB.

It is a commonly occurring problem, though I am not aware of it occurring on every computer.  There is no way to tell in advance whether a particular computer / computing device will experience the problem.

GaoZiHong, 2020-11-18:
thank you. In this case, when I only open RGB in one window, RGB cannot be used, right? I have to open RGB in one window and activate the motion module in another window. Is my description correct?

发自我的iPhone

MartyG-RealSense, 2020-11-18:
Basically, yes.  You can use RGB and depth in the same Viewer window, but do not activate the motion module in that same window.

GaoZiHong, 2020-11-18:
thank you. I will verify this result tomorrow morning, I hope it is useful. I would like to know the possible reasons why RGB images can not be used in a single window but depth maps can be used? What exactly is the function of the motion module? Why is the above operation correct in two windows? Does this problem only occur under the arm architecture of TX2 or is it possible under the x86 architecture?

发自我的iPhone

MartyG-RealSense, 2020-11-18:
You can use both RGB and Depth in the same Viewer session.  The trigger for the problem with the RGB is always the activation of the motion module.

So there should be no problem with having both RGB and depth on at the same time in the same Viewer session so long as you do not turn the motion module on too.

The motion module displays gyroscope and acceleration data from the IMU component (for example, how fast the camera is moving and if it is tilted at an angle).  If you do not need to use that data then you do not need to activate the motion module.  It is not needed for depth and RGB information.

GaoZiHong, 2020-11-19:
Thank you, I went to test the method you mentioned this morning, unfortunately the method did not work. I still cannot use an RGB camera. After I open the motion module, the global time buffer error is displayed. What are the other possible solutions?

MartyG-RealSense, 2020-11-19:
The easiest way to resolve this problem is to not open the motion module.  It is not needed for streamng RGB or depth.

MartyG-RealSense, 2020-11-26:
Hi @GaoZiHong   Do you require further assistance with this case, please?    Thanks!

GaoZiHong, 2020-11-26:
yes,please

发自我的iPhone

MartyG-RealSense, 2020-11-26:
I ran further tests with the new 2.40.0 Viewer and firmware 5.12.9.0 and the RGB problem after activating the Motion Module is still present.  I also found that the problem was unrelated to the camera.   If a D435i was attached when the RGB had No Frames Received and another camera model such as the D415 was used instead, the RGB still could not receive frames.  Only a reboot of the computer corrected the problem.

These results confirm that there is nothing that can be done right now to correct the problem in the Viewer, unfortunately.  If you need motion data, you will have to use **only** the Motion Module in the Viewer.  

The **rs-multicam** program that can show depth, RGB and IMU simultaneously from a single D435i is not affected by this RGB issue.

https://github.com/IntelRealSense/librealsense/tree/master/examples/multicam

[image]

GaoZiHong, 2020-12-03:
No thanks. This question is not so important to me now, I am now using x86 architecture equipment. If I still encounter this problem when I use TX2 in the future, I will ask for help again. Thanks again

MartyG-RealSense, 2020-12-03:
Thanks so much for the update @GaoZiHong - I will close this case now.

closed 2020-12-03 by MartyG-RealSense