realsenseai/librealsense #7778: Error During CMake Build that yields: "Configuring incomplete, errors occurred!"

ahmed040, 2020-11-16:
### Issue Description

*** Please note I am new to this whole procedure and to the community so please forgive any inconvenience from my side or any procedure I mistakenly neglected ***

First of all I would like to explain what I am trying to do. I am trying to generate the realsense2.sln file from the librealsense C# wrapper. I followed the procedure in the readme file in the csharp wrapper directory "librealsense/wrappers/csharp/readme.md"
till the step of cmake command "cmake .. -DBUILD_CSHARP_BINDINGS=ON -DBUILD_SHARED_LIBS=ON" When I entered the command on the Visual Studio Developer Command Prompt the build did not succeed as there was a list of errors. I copied the build log and error messages from the prompt to a file and attached it with my issue as you will find. If anyone here has seen this error before or knows what exactly I did wrong please help.
[Visual Studio Command Prompt Response for CMake Build.txt](https://github.com/IntelRealSense/librealsense/files/5545017/Visual.Studio.Command.Prompt.Response.for.CMake.Build.txt)

MartyG-RealSense, 2020-11-16:
Hi @ahmed040   If you are new to wrapper building, you may find useful the user-friendly librealsense C# wrapper setup guide at te link below:

https://cccommunications.com/intel-realsense-3d-depth-camera-sdk-setup-for-c-development/

Also, if you are using Windows then you can find pre-built DLLs for realsense2.dll and Intel.RealSense.dll at the folder location below:

C:\Program Files (x86)\Intel RealSense SDK 2.0\bin\x64

[image]

dorodnic, 2020-11-16:
Hi @ahmed040 
I second @MartyG-RealSense recommendations. 
In addition, CMake is failing because it cannot find git installed on the system.

ahmed040, 2020-11-17:
It turned out that git was not installed and that was the root cause of the problem. Thanks everyone

closed 2020-11-17 by ahmed040

MartyG-RealSense, 2020-11-17:
Thanks very much @ahmed040  for the update - great to hear that the advice of @dorodnic worked for you    :)