realsenseai/librealsense #7780: intel realsense d435 depth stream is noisy. what would the problem?

Tirtha-Bidhayok, 2020-11-16:
| | |
|---|---|
|**librealsense**|2.31.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435|
|**Serial Number**|938422072894|
|**Firmware Version**|05.12.01.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descr
[image]
iptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|939423020839|
|**Firmware Update Id**|939423020839|

Please provide a description of the problem

MartyG-RealSense, 2020-11-16:
Hi @Tirtha-Bidhayok   How far away is the camera from the surface that is being observed, please?   The D435 has a default minimum depth sensing distance of 0.1 meters (10 centimeters), so the image may start to break up if the camera is closer to the observed area than that.

Below is an image I produced for comparison by moving a D435i close to a wall at a distance of less than 10 cm:

[image]

Tirtha-Bidhayok, 2020-11-16:
> Hi @Tirtha-Bidhayok My first question would be to ask how far away the camera is from the surface that is being observed, please? The D435 has a default minimum depth sensing distance of 0.1 meters (10 centimeters), so the image may start to break up if the camera is closer to the observed area than that.
> 
> Below is an image I produced for comparison by moving a D435i close to a wall at a distance of less than 10 cm:
> 
> [image]
> 
> If you are depth sensing at very close range, you can reduce the minimum distance of the camera and enable it to get closer to a surface by increasing the value of the **Disparity Shift** option.

Tirtha-Bidhayok, 2020-11-16:
> Hi @Tirtha-Bidhayok My first question would be to ask how far away the camera is from the surface that is being observed, please? The D435 has a default minimum depth sensing distance of 0.1 meters [...]

I've tried 1m, 0.7m, 0.4m distance from the camera. but the stream was noisy.

MartyG-RealSense, 2020-11-16:
Have you been able to successfully stream depth with the camera before now, please?

If the camera is further than the minimum distance away from the observed surface, then mis-calibration of the camera could also result in a broken image like the one in your opening comment.  This can be corrected by performing a recalibration of the camera using the **Dynamic Calibrator** tool for the 400 Series cameras.

https://downloadcenter.intel.com/download/29618/Intel-RealSense-D400-Series-Dynamic-Calibration-Tool

Tirtha-Bidhayok, 2020-11-16:
> Have you been able to successfully stream depth with the camera before now, please?
>
> If the camera is further than the minimum distance away from the observed surface, then mis-calibration of the [...]

i have completed a big project with this camera 5 months ago. On that time the camera performed correctly. After the project i have put it down camera in the box safely. But today when ran this camera for another project he problem happened. 
Two sample image has been added before 5 months ago
[image]
[image]

, and today's.

MartyG-RealSense, 2020-11-16:
Excessive temperature may cause mis-calibration.  Can you confirm please that the camera was stored in a location for 5 months that would not have exceeded the recommended long term storage conditions in the table below from the 400 Series data sheet document, please?

[image]

Chandler, 2020-11-19:
@Tirtha-Bidhayok I would also confirm that the IR projector is on by viewing the IR stream in realsense-viewer and ensuring you can see the dot pattern. Almost looks like you have no IR.

MartyG-RealSense, 2020-12-03:
Case closed due to no further comments received.

closed 2020-12-03 by MartyG-RealSense

Oldog, 2020-12-29:
Sometimes I come up with the same problem. But everything is OK after resetting the camera hardware.

MartyG-RealSense, 2020-12-29:
Thanks very much @Oldog for your advice to @Tirtha-Bidhayok