realsenseai/librealsense #7783: Unlock Intel Realsense D455 for Updates

pandurdo, 2020-11-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | 05.08.15.00 | 
| Operating System & Version | {Win 10} | 
| Platform                                 | PC |
| SDK Version                            | v2.39.0 | 

### Issue Description

After trying to update my Intel Realsense D455 to the newest update, I got the message that the device is locked for update.
Is there a way to unlock the camera?

MartyG-RealSense, 2020-11-16:
Hi @pandurdo   Could you open the RealSense Viewer program please and use the 'Info' option to list the details of your camera to confirm which firmware version is currently installed.   It should not be the ' 05.08.15.00' firmware listed in the Required Info details at the top of this case, as that was one of the first camera firmwares dating back to December 2017.  Cameras are shipped with at least the minimum firmware that supports them, and I believe for D455 that was a recent firmware from this year (2020).

[image]

What method are you using to update your camera firmware?

pandurdo, 2020-11-16:
[image]

I tried to update it with the viewer as well as with the in SDK included tool 'rs-fw-update.exe' and the manually downloaded firmware '05.12.08.200' and got messages like:

"16/11 12:51:54,383 ERROR [18572] (types.h:313) hr returned: HResult 0x8007001f: "Ein an das System angeschlossenes Ger├ñt funktioniert nicht."
 16/11 12:51:55,558 ERROR [19380] (handle-winusb.h:55) CreateFile failed, error: 5
 16/11 12:51:55,560 ERROR [19380] (context.cpp:431) Exception thrown from user callback handler

failed to locate a device in FW update mode"

When I now try to update to this firmware, I get: 

"firmware update started

firmware update progress: 8[%]RealSense error calling rs2_update_firmware_cpp(device:000001F8CF8BAF20, fw_image:000001F8D044F060):
    Device: xxxxxxxxxxxx is locked for update.
Use firmware version higher than: 5.12.8.200"

MartyG-RealSense, 2020-11-16:
Okay, I see you have a **D435**.  The details at the top of this case said D455.  Thank you for the information.  So it would make sense why you might have an early firmware like **05.08.15.00** installed by default after purchasing the camera, as D435 was one of the original launch models of the 400 Series range.

It looks as though the English version of the HResult error logged above is **A device attached to the system is not functioning**.  

Could you check the Windows privacy settings for cameras please to see whether Windows is set to allow applications to access the camera using the instructions in the link below?

https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

pandurdo, 2020-11-16:
Actually it is a **D455**, I don't know why it is recognized as D435. I tried to update it on a different PC, where it was also detected as D435. After successfully updating it to the newest firmware, it is recognized as **D455**. But after switching back to my working PC, Windows downgrades the firmware back to '05.08.15.00'.

As I told you, **I am no more able to update this camera (on any PC)** due to the fact, that it is **locked for updates (see error above).**

MartyG-RealSense, 2020-11-16:
How many times have you attempted to update the firmware of the camera, please?  RealSense cameras have a limit of 20 downgrades (installing a firmware version lower than the currently installed one).  After that point, you can only upgrade to a firmware version newer than the currently installed version.  

So if the installed firmware was being recognised as **5.12.8.200** and the downgrade limit had been reached, I would expect to be told "Use firmware version higher than: 5.12.8.200" each time that you tried to install 5.12.8.200 or older.

.Let's try to force a firmware update in the **recovery** mode of rs-fw-update.  First, please download the firmware .bin file for 5.12.8.200 if you have not done so already.

https://dev.intelrealsense.com/docs/firmware-releases

[image]

Place the firmware bin file in the same folder location as rs-fw-update and then launch rs-fw-update in recovery mode using the instruction below:

**rs-fw-update -r -f Signed_Image_UVC_5_12_8_200.bin**

pandurdo, 2020-11-16:
Here the resulting message:

rs-fw-update.exe -r -f Signed_Image_UVC_5_12_8_200.bin

update to FW: Signed_Image_UVC_5_12_8_200.bin

recovering device:
Name: Intel RealSense D4xx Recovery, serial number: unknown, update serial number: 025523050887, firmware version: unknown, USB type: unknown

firmware update started

firmware update progress: 8[%]
failed to recover device

MartyG-RealSense, 2020-11-16:
I notice that the USB type of your camera connection is not being displayed in the RealSense Viewer information.  It is usually beside the camera name and says '2.1' if it is a USB connection and '3.2' if a USB 3 connection.   The image below indicates where the USB type should be displayed on your connection:

[image]

And here is what my D435i camera's information looks like:

[image]

Can you confirm please whether your camera information remains the same if you go to the **More** option and select the **Hardware Reset** menu option.

pandurdo, 2020-11-16:
Nothing happens by selecting **Hardware Reset**.

[image]

MartyG-RealSense, 2020-11-16:
I have not seen a case before where the USB type is not detected / displayed at all.   I will consult within Intel about it.   I apologise for the delay in the meantime.

dorodnic, 2020-11-16:
Having the device locked for updates is simply a function of too many downgrade-upgrade cycles. 
5.8.15 on a D455 unit is not going to do anything good, since this firmware predates D455 hardware by 3 years... 
I'm very concerned about this - "Windows downgrades the firmware back to '05.08.15.00'." Could you please provide more details, such as exact Windows version?

pandurdo, 2020-11-16:
I did an successful upgrade to 05.12.08.200 one a Windows 10 PC, where such cameras were never used. When I plugged in the cam on my working PC with Windows 10 up-to-date (where I used Intel Realsenses such as D435 and D435i), the D455 was detected and again downgraded to 05.08.15.00. And on this PC I was not able to upgrade neither via Viewer nor by the rs-fw-update.exe.

MartyG-RealSense, 2020-11-23:
Hi @pandurdo   Do you require further assistance with this case, please?   Thanks!

pandurdo, 2020-11-23:
Yes, Since my problem is not solved yet. I have a camera,which is not working and is not updatable.

MartyG-RealSense, 2020-11-24:
@dorodnic   Do you have further advice about the problem of @pandurdo that you commented on earlier, please?  Thanks!

MartyG-RealSense, 2020-12-01:
Hi again @pandurdo   Your previous problem was that your camera would only allow you to upgrade to a firmware newer than **5.12.8.200**.   As the new firmware **5.12.9.0** is now available, does your camera allow you to update to it?

pandurdo, 2020-12-03:
Yes, upgraded it and it works now.

MartyG-RealSense, 2020-12-03:
Great to hear that news @pandurdo

pandurdo, 2020-12-10:
No, thank you!

closed 2020-12-10 by pandurdo

MartyG-RealSense, 2020-12-10:
Excellent, thanks for the update!   :)