realsenseai/librealsense #7786: Is there a way to align the captured frames in Top side front orthogonal views

readblack1234, 2020-11-16:
|librealsense|2.35.2 RELEASE|
|OS|Linux|
|Name|Intel RealSense SR305|
|Serial Number|636204001701|
|Firmware Version|3.27.3.0|
|Camera Locked|YES|
|Product Line|SR300|
|Asic Serial Number|636204001701|
|Firmware Update Id|636204001701|                       | 

### Issue Description
I use rs pipe to get frames captured.
      auto frames = pipe.wait_for_frames();
      auto depth = frames.get_depth_frame();

How do I get orientation of the captured frames in certain way like front top side view?
I would visualize these in pcl point cloud and there I can just view captured as is would be one way.
If there were some parameters passed to orient the object with respect to camera in front top or side view, that would be easiest.

MartyG-RealSense, 2020-11-16:
Hi @readblack1234   If you want to generate a PCL point cloud and rotate it, the SDK's PCL wrapper has an example called **rs-pcl-color** that references a point cloud rotation function for the viewer-object that visualizes the cloud.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/pcl/pcl-color
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/pcl/pcl-color/rs-pcl-color.cpp#L276

If use of PCL is not vital, the SDK's C++ example **rs-post-processing** also offers code for point cloud rotation.

https://github.com/IntelRealSense/librealsense/blob/master/examples/post-processing/readme.md

readblack1234, 2020-11-21:
Thanks @MartyG-RealSense . The example of rs-color is useful. I have not see any orientation code in there though, it is generating a point cloud in pcl with both depth and color. 
I did found two issues or things I had to modify to get point cloud
1. Somehow it enable and disable the laser, so I had to comment out line
//depth_sensor.set_option(RS2_OPTION_LASER_POWER, 0.f); // Disable laser
2. The point cloud generated was only showing edge and it was difficult to make it look like 3d cloud. For that I had to turn this true,
    cloud->is_dense = true;
[image]

Now I can generate color pcl, that is useful. however, I am looking how can i have orthogonal alignment for these easily done? Also, the mouse control to zoom in results in very large jump, I use + to zoom in but still wonder if there is better mouse control?

MartyG-RealSense, 2020-11-21:
My thinking was that if the point cloud could be rotated then it might be possible to program controls to teleport to a viewpoint instead of rotating progressively from one position to another.

I would think that the mouse sensitivity is related to the input settings in the operating system of the specific computer rather than something that can be set in the program.

closed 2020-12-05 by MartyG-RealSense