realsenseai/librealsense #7787: Native Crash In Unity Editor (Dragging & Dropping Crash)

ababilinski, 2020-11-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415, D435, D455 | 
| Firmware Version                   | 05.12.06.00 | 
| Operating System & Version |  Win 10  | 
| Kernel Version (Linux Only)    | NA                                       | 
| Platform                                 | Unity  |
| SDK Version                            |  [v2.39.0](https://github.com/IntelRealSense/librealsense/releases/tag/v2.39.0)                         | 
| Language                            |  Unity/C#                       | 
| Segment			|  NA                          | 

### Issue Description
Unity Crashes due to native error in the precompiled plugins in the [Realsense Wrapper](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unity). This issue can be reproduced by following these steps: 

1. Open a **Unity Project**
2. Import Unity Wrapper `.unitypackage`
3. Play the example scene(s)
4. Repeat step 3 four or five times
5. Drag an object from Library or Hierarchy in the Editor
6. Observer crash

## Resolution Description
The issue can resolved by following these steps: 
1. Compile the `Intel.RealSense.dll` & `realsense2.dll` from the current master branch 
2. replace the plugins that were imported with the `.unitypackage` with the ones produced in step 1

### Fixed Plugin Files
Attached are the re-compiled library files from commit [[d8f5d42]](https://github.com/IntelRealSense/librealsense/commit/d8f5d4212df85522a14b4a7b83bf4d54219b06fa) 

To replicate the fix, download the zip and replace the plugin folder located `Assets/RealSenseSDK2.0/Plugins/`
**Fixed Plugin Folder:** [Plugins.zip](https://github.com/IntelRealSense/librealsense/files/5548513/Plugins.zip)

MartyG-RealSense, 2020-11-17:
Hi @ababilinski   Thanks very much for your detailed report!

I would add that if other Windows users experience this issue and want to try your guide, they can obtain pre-built versions of Intel.RealSense.dll and realsense2.dll from the SDK folder instead of compiling the DLLs themselves.

https://github.com/IntelRealSense/librealsense/issues/7778#issuecomment-728163719

MartyG-RealSense, 2020-11-26:
Hi @ababilinski   Do you require further assistance with this case, please?  Thanks!

ababilinski, 2020-11-27:
@MartyG-RealSense no, I just wanted to post this for anyone else that's having the same bug as I am.

MartyG-RealSense, 2020-11-27:
Great, thanks very much for adding the information to the RealSense community!   I will close this case now.   Your information will remain findable with a web search.   Thanks again   :)

closed 2020-11-27 by MartyG-RealSense