realsenseai/librealsense #7788: On-chip calibration from C#

joshberry, 2020-11-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.12.06.00 | 
| Operating System & Version |   Win 10  |                                         | 
| Platform                                 | PC  |                    | 
| Language                            |  C#                         | 

### Issue Description
Are the on-chip self calibration functions available from the C# interface? I've looked through the C# wrapper source code and am not seeing them (appears they've been exposed in the python wrapper though).

Assuming they are not yet implemented in the C# wrapper, are there any plans to do so? 

I've attempted to PInvoke the functions directly, but am running into errors. For example, here's my attempt to expose the rs2_get_calibration_table function from C#.

```
[DllImport(RS_LIB, CallingConvention = CallingConvention.Cdecl)]
internal static extern IntPtr rs2_get_calibration_table(IntPtr device, [MarshalAs(UnmanagedType.CustomMarshaler, MarshalTypeRef = typeof(ErrorMarshaler))] out object error);
```

```
// Calling it later...
object error;
var calibTable = rs2_get_calibration_table(dev.Handle, out error);
```

When I try to call it from my code I get the following error:

> System.Exception: 'Object does not support "librealsense::auto_calibrated_interface" interface! '

It seems like this should work as I've setup PInvoke just like it's done within the C# wrapper for the rs2_create_flash_backup function which has a similar signature.

Any tips or advice on how to correctly use the on-chip calibration functions from C# would be much appreciated. Thanks!

RealSenseSupport, 2020-12-04:
I think you use a wrong device handle in your code.

Please review the code.
Or try https://github.com/IntelRealSense/librealsense/pull/7915

RealSenseSupport, 2020-12-17:
@joshberry PR https://github.com/IntelRealSense/librealsense/pull/7915 now merged to development branch. Could you please have a try? Thanks!

RealSenseSupport, 2020-12-28:
@joshberry SDK v2.41 now released and includes the PR#7915. Please have a try. Thanks!

joshberry, 2020-12-28:
@RealSenseSupport Thanks, I'll give it a try...

RealSenseSupport, 2021-01-15:
@joshberry Did you get chance to try and any update? Thanks!

joshberry, 2021-01-15:
@RealSenseSupport I have not, but should be circling back to it later this month. I'll post here once I've had a chance to use the new C# interface.

joshberry, 2021-03-04:
@RealSenseSupport I'm finally getting around to implementing the calibration functions in our app and am running into a problem when using the C# wrapper. Every time I call the RunOnChipCalibration method, it returns the following error:

```
System.Exception: 'Calibration didn't converge! (EDGE_TO_CLOSE)
Please retry in different lighting conditions'
```

The method call I'm making seems pretty straightforward:
```
var calibrationTable = calibratedDevice.RunOnChipCalibration("{\"speed\": 3, \"scan_parameter\": 0, \"data sampling\": 0}", out health, 10000);
```

I've tried various combinations of the input parameters with the same result. I have the rs camera on a tripod and am able to successfully complete the calibration from rs viewer, so I don't think it's the camera placement or lighting conditions causing this as the error suggests.

Any idea what I'm doing wrong here?

agrunnet, 2021-03-04:
Are you using 256x144 resolution?

joshberry, 2021-03-05:
That appears to have been my problem. Must have missed that part in the whitepaper. Thanks @agrunnet!

joshberry, 2021-03-05:
@RealSenseSupport This appears to work great for performing on-chip calibration through the C# wrapper. Thanks!

I recently became aware of a new feature related to tare calibration which automatically determines the ground truth based on a pre-defined target instead of requiring the user to provide a known distance. Is this available through the C# wrapper as well? I'm not seeing it, so just wondering if I'm missing something or if it's planned for the future. Appears to have been added to the base library on pr #7535 and released on 2.41 with the addition of on-chip calibration to the C# wrapper.

RealSenseSupport, 2021-03-18:
@joshberry The new feature of calculating the tare ground with the specific target is not available in C# wrapper yet. It needs the specific target which is not released yet. So currently please do the tare calibration by inputting ground truth to go ahead. Thanks!

RealSenseSupport, 2021-03-31:
@joshberry Any other questions about this ticket? Looking forward to your update. Thanks!

joshberry, 2021-03-31:
@RealSenseSupport Do you know when tare ground truth calculation based on a the pre-defined target will be added to the C# wrapper? For our use case, it's not possible to input the ground truth manually, so we'll need to wait until that's available.

RealSenseSupport, 2021-04-02:
@joshberry You could configure one default value for the ground truth and put the camera at such distance to go ahead. In this case, do you still need the way using the pre-defined target for C# wrapper? Looking forward to your reply. Thanks!

joshberry, 2021-04-02:
@RealSenseSupport The customer I'm working with doesn't want to have to define a ground truth for a variety of reasons. Their Intel account manager provided the targets and advised this approach. Do you know when the functionality will be exposed through the C# wrapper?

MartyG-RealSense, 2026-10-23:
Case closed due to no further comments received after a significant time period.