realsenseai/librealsense #7789: Unity wrapper not playing back recordings from Intel RealSense Viewer

dbarnes10, 2020-11-17:
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 / D435i | 
| Firmware Version                   |05.12.05.00 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |  2.0 (not sure how to find this in the Unity subsection)                     | 
| Language                            |  unity                          | 
| Segment			|  creative arts                          | 

### Issue Description
".bag" files recorded using the Intel RealSense Viewer play back correctly in the viewer, but cause an error in the Unity wrapper. See below for error output. Experimentation shows the error occurs when the PointCloud (RsProcessingProfile) is being used to generate a point cloud. The prior application of  Filters (RsProcessingProfile) issues no errors when used alone. But I'm pretty keen for some visual output.

This error seems independent of the .bag files I used.  It is basically using the "Simple" test scene included in the unity wrapper, and setting "Playback" as the mode of RsDevice, and selecting an existing .bag file on disk.

Noting I cannot upgrade my 435/435i firmware, as these are recordings done on the firmware listed above.

Thanks in advance!

InvalidOperationException: Error while running PointCloud processing block, check the log for info
Intel.RealSense.ProcessingBlock.Process[T] (Intel.RealSense.Frame original) (at <39ff202c67d64a63acbba4ae542e7167>:0)
Intel.RealSense.ProcessingBlock.Process (Intel.RealSense.Frame original) (at <39ff202c67d64a63acbba4ae542e7167>:0)
RsPointCloud.Process (Intel.RealSense.Frame frame, Intel.RealSense.FrameSource frameSource) (at C:/Users/dbarnes/Documents/Unity/intelRS-projection-02/Assets/RealSenseSDK2.0/Scripts/ProcessingBlocks/RsPointCloud.cs:59)
RsProcessingPipe.ProcessFrame (Intel.RealSense.Frame frame, Intel.RealSense.FrameSource src) (at C:/Users/dbarnes/Documents/Unity/intelRS-projection-02/Assets/RealSenseSDK2.0/Scripts/RsProcessingPipe.cs:99)
UnityEngine.Debug:LogException(Exception)
RsProcessingPipe:ProcessFrame(Frame, FrameSource) (at C:/Users/dbarnes/Documents/Unity/intelRS-projection-02/Assets/RealSenseSDK2.0/Scripts/RsProcessingPipe.cs:119)
Intel.RealSense.CustomProcessingBlock:ProcessingBlockCallback(IntPtr, IntPtr, IntPtr)
Intel.RealSense.NativeMethods:rs2_process_frame(IntPtr, IntPtr, Object&)
Intel.RealSense.CustomProcessingBlock:Process(Frame)
RsProcessingPipe:OnFrame(Frame) (at C:/Users/dbarnes/Documents/Unity/intelRS-projection-02/Assets/RealSenseSDK2.0/Scripts/RsProcessingPipe.cs:68)
Intel.RealSense.CustomProcessingBlock:ProcessingBlockFrameCallback(IntPtr, IntPtr)
Intel.RealSense.NativeMethods:rs2_synthetic_frame_ready(IntPtr, IntPtr, Object&)
Intel.RealSense.FrameSource:FrameReady(Frame)
RsProcessingPipe:ProcessFrame(Frame, FrameSource) (at C:/Users/dbarnes/Documents/Unity/intelRS-projection-02/Assets/RealSenseSDK2.0/Scripts/RsProcessingPipe.cs:112)
Intel.RealSense.CustomProcessingBlock:ProcessingBlockCallback(IntPtr, IntPtr, IntPtr)
Intel.RealSense.NativeMethods:rs2_process_frame(IntPtr, IntPtr, Object&)
Intel.RealSense.CustomProcessingBlock:Process(Frame)
RsDevice:RaiseSampleEvent(Frame) (at C:/Users/dbarnes/Documents/Unity/intelRS-projection-02/Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:148)
RsDevice:WaitForFrames() (at C:/Users/dbarnes/Documents/Unity/intelRS-projection-02/Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:160)
System.Threading.ThreadHelper:ThreadStart()

MartyG-RealSense, 2020-11-17:
Hi @dbarnes10   As a starting point, may I first ask whether you are using Unity 2019 (e.g 2019.4) or Unity 2020?  Results when using Unity 2020 cannot be guaranteed.  For example, there have been a smal number of reports with issues related to **ARBackgroundRenderer** in Unity 2020.

dbarnes10, 2020-11-18:
Hey thanks Marty. I updated my Unity project to 2019.4 (LTS) and I rebuilt librealsense from scratch using Visual Studio 2019, and hey presto, it works!  Thanks for the clues ;-)

closed 2020-11-18 by dbarnes10

MartyG-RealSense, 2020-11-18:
Great to hear @dbarnes - thanks very much for the update about your success!

MartyG-RealSense, 2020-11-18:
Hi again @dbarnes10   SDK 2.40.0 released today adds Unity fixes.

https://github.com/IntelRealSense/librealsense/pull/7625