realsenseai/librealsense #7790: Extender Cable Recomendation for the Lidar 515 Camaera

TommasoBendinelli, 2020-11-17:
I need to place the L515 Camera around 5 meters away from my PC. Hence I need an extender. I have bought this cable DELOCK 83089 https://www.delock.de/produkte/G_83089/merkmale.html?setLanguage=en, but the Camera, when plugged through the extender, is not working (Windows see the Camera, but cannot find a proper driver).
When the camera is attached directly, everything is working fine.

Chandler, 2020-11-19:
You cannot extend USB 3.0 to 5 meters after 3 meters I believe it will fall back to usb 2.0.

> t[he maximum practical length is 3 meters (9.8 ft)](https://en.wikipedia.org/wiki/USB_3.0#:~:text=Unlike%20previous%20standards%2C%20the%20USB,3%20meters%20(9.8%20ft).)

Make sure you are running the latest L515 firmware since it was suggested that it may work with USB 2.0 which would allow you to use a longer distance -- but in general be aware that the USB 3.0 length limit is a major limiting factor.

https://github.com/IntelRealSense/librealsense/issues/7552

TommasoBendinelli, 2020-11-23:
Do you suggest any active cable for longer distances?

cdb0y511, 2020-11-26:
> Do you suggest any active cable for longer distances?

Hi, @TommasoBendinelli 
Actually, there is a way to extend it to almost more than 100m.
You need a fiber cable, receiver, and USB3 to fiber adapter. They may cost more than 300 bucks. Be aware, the receiver needs an additional power supply.

RealSenseSupport, 2021-01-08:
Hi @TommasoBendinelli 

Hopefully you found a working cable.  We cannot provide such a recommendation.

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-16 by RealSenseSupport