realsenseai/librealsense #7792: why my D455 can't\

AmazingSealock, 2020-11-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2020-11-17:
Hi @AmazingSealock   If you are observing the set of six plates in a black framework, it is possible that the camera is interpreting it as a **repetitive pattern**.  Having horizontal or vertical repeating patterns in a scene (such as window blinds or a row of fence posts) can confuse the camera.  Turning the camera may disrupt that repetitive pattern.

The subject is discussed in the link below.

https://github.com/IntelRealSense/librealsense/issues/6713

Within that case, links to resources about methods that may help to negate the effect are provided.

https://github.com/IntelRealSense/librealsense/issues/6713#issuecomment-651114720

Repetitive patterns are also discussed in the link below.

https://github.com/IntelRealSense/librealsense/issues/4082

AmazingSealock, 2020-11-20:
Thank you @MartyG-RealSense, I've now switched to looking for a **bucket** in the depth Image, but there's still a problem. It doesn't look like the repetitive pattern caused this. 
I've already calibrated my D455.

MartyG-RealSense, 2020-11-20:
For the D455, the recommended optimal depth accuracy resolution is **848x480** rather than the 1280x720 in your Viewer image above.   Do your results improve if you use 848x480 please?

MartyG-RealSense, 2020-12-03:
Case closed due to no further comments received.

closed 2020-12-03 by MartyG-RealSense

AmazingSealock, 2020-12-03:
Thank you, I sovlede this problem!

MartyG-RealSense, 2020-12-03:
Great to hear - thanks for the update   :)