realsenseai/librealsense #7793: fps limitation on Jetson Xavier 70 FPS? d455

AndreV84, 2020-11-17:
related thread
https://forums.developer.nvidia.com/t/realsense-d455-cannot-get-90fps-video-frames-on-the-xaiver-agx/159623/4
any workarounds to increase fps?

tedtman, 2020-11-18:
Hello @AndreV84 have you encountered this issue or just noting the NVIDIA posting? If you have observed this, what is the particular configuration you are trying to run. I have streamed 848 x 480 @90fps using realsense-viewer with a single D455.

AndreV84, 2020-11-18:
Hi @tedtman 
Thank you for your response.
I tried to look into the issue.
Did you achieve @90 fps on x86_64 or of Nvidia Jetson device?
I just tried with d455 on host PC x86_64 and got low fps also 
`v4l2-ctl -d /dev/video4 --set-fmt-video=width=1280,height=720,pixelformat=UYVY --stream-count=1000 --stream-mmap
<<<<<<<<<<<<<<< 13.72 fps
<<<<<<<<< 10.95 fps
<<<<<<<< 10.19 fps
<<<<<<<<< 9.75 fps
<<<<<<<< 9.50 fps
<<<<<<<< 9.22 fps
<<<<<<<< 8.98 fps
<<<<<<< 8.83 fps
<<<<<<<< 8.68 fps`
I can try on Jetson later
right now we got it  connected to x86_64

FrankCreen, 2020-11-18:
> Hello @AndreV84 have you encountered this issue or just noting the NVIDIA posting? If you have observed this, what is the particular configuration you are trying to run. I have streamed 848 x 480 @90fps using realsense-viewer with a single D455.

With the latest librealsense-2.39.0 and  fireware 5.12.8.20, I can get the 90FPS in the realsense-viewer with a 480*270 resolution on the Jetson Xaiver Device. But I can't do this with the C++ code. The code is below:

```
// include the librealsense C++ header file
#include <librealsense2/rs.hpp>

#include <opencv2/opencv.hpp>

#include <chrono>

using namespace std;
using namespace cv;

int main()
try
{
    //Contruct a pipeline which abstracts the device
    rs2::pipeline pipe;

    //Create a configuration for configuring the pipeline with a non default profile
    rs2::config cfg;
    //Add desired streams to configuration
    cfg.enable_stream(RS2_STREAM_COLOR, 480, 270, RS2_FORMAT_BGR8, 90);

    //Instruct pipeline to start streaming with the requested configuration
    pipe.start(cfg);

//Sensor Settings
    //Depth Camera Settigs
    auto sensor11 = pipe.get_active_profile().get_device().query_sensors()[0];
    sensor11.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);

    //RGB Camera Settigs
    auto sensor2 = pipe.get_active_profile().get_device().query_sensors()[1];
    sensor2.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);
    sensor2.set_option(RS2_OPTION_AUTO_EXPOSURE_PRIORITY, 0);
    sensor2.set_option(RS2_OPTION_ENABLE_AUTO_WHITE_BALANCE, 1);
    //Controls Settings
    sensor2.set_option(RS2_OPTION_BRIGHTNESS, 0.0);
    sensor2.set_option(RS2_OPTION_CONTRAST, 50.0);
    sensor2.set_option(RS2_OPTION_EXPOSURE, 160.0);
    // sensor2.set_option(RS2_OPTION_GAIN,64.0);
    sensor2.set_option(RS2_OPTION_GAMMA, 300.0);
    sensor2.set_option(RS2_OPTION_HUE, 0.0);
    sensor2.set_option(RS2_OPTION_SATURATION, 64.0);
    sensor2.set_option(RS2_OPTION_SHARPNESS, 50.0);
    sensor2.set_option(RS2_OPTION_WHITE_BALANCE, 4600.0);

    sensor2.set_option(RS2_OPTION_POWER_LINE_FREQUENCY, 0);
    sensor2.set_option(RS2_OPTION_GLOBAL_TIME_ENABLED, 1);

    // Display in a GUI
    const auto window_name = "Display Video";
    namedWindow(window_name, WINDOW_AUTOSIZE);

    //用于计时
    //show time
    auto start = chrono::system_clock::now();
    int frame = 0;

    while (waitKey(1) < 0 && getWindowProperty(window_name, WND_PROP_AUTOSIZE) >= 0)
    {
        rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera

        //获取彩色图像帧
        rs2::frame color = data.get_color_frame();

        //保存彩色图像
        Mat image(Size(480, 270), CV_8UC3, (void *)color.get_data(), Mat::AUTO_STEP);

        // Update the window with new data
        imshow(window_name, image);

        frame++;
        //one loop end
        auto end = chrono::system_clock::now();
        auto duration = chrono::duration_cast<chrono::microseconds>(end - start);
        double timecost = double(duration.count()) * chrono::microseconds::period::num / chrono::microseconds::period::den;

        cout << frame / timecost << " FPS" << endl;
    }

    return 0;
}
catch (Exception ex)
{
    cout << "Error!!!" << endl;
    cerr << ex.msg << endl;
}
```

tedtman, 2020-11-18:
@AndreV84 I have run D455 depth 848x480 16 bit 90 fps on both NVDIA AGX Jetson and x86_64 using realsense-viewer from the SDK.

AndreV84, 2020-11-19:
solved
Thank you everyone

closed 2020-11-19 by AndreV84