realsenseai/librealsense #7797: failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS in realsense-viewer

viisautta, 2020-11-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.08.200 | 
| Operating System & Version |  Fedora33  | 
| Kernel Version (Linux Only)    |  5.9.8-200.fc33.x86_64 | 
| Platform                                 | laptop |
| SDK Version                            | 2.39.0 | 
| Language                            | C++ | 
| Segment			|  | 

### Issue Description
$ realsense-viewer 
 18/11 10:05:31,970 ERROR [140416875738688] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 18/11 10:05:31,970 WARNING [140417197844032] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Enable / disable backlight compensation
 18/11 10:05:31,970 WARNING [140417197844032] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image brightness
 18/11 10:05:31,970 WARNING [140417197844032] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image contrast
 18/11 10:05:31,970 WARNING [140417197844032] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Controls exposure time of color camera. Setting any value will disable auto exposure
 18/11 10:05:31,970 WARNING [140417197844032] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image gamma setting
 18/11 10:05:31,970 WARNING [140417197844032] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image hue
 18/11 10:05:31,970 WARNING [140417197844032] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image saturation setting
 18/11 10:05:31,970 WARNING [140417197844032] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image sharpness setting
 18/11 10:05:31,970 WARNING [140417197844032] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
Segmentation fault (core dumped)

$ ls /etc/udev/rules.d/
557-primesense-usb.rules  70-persistent-ipoib.rules  99-realsense-libusb.rules

it can reproduce after plug out and in sensor

MartyG-RealSense, 2020-11-18:
Hi @viisautta   Let's deal with the **RS2_USB_STATUS_ACCESS** error and the **Exception was thrown when inspecting Raw RGB Camera property** warnings separately, starting with the raw RGB warnings first.

Could you try launching the Viewer without the D435 plugged in please and see whether you still get the RGB warnings.  This warning may occur on some computers if a non-RealSense UVC camera such as a USB webcam or a built-in laptop webcam is detected.   If there is a non-RealSense cam, unplugging it from USB or disabling a built-in one in system settings may stop the warnings from appearing.

viisautta, 2020-11-18:
start realsense_viewer normal with unplugging this D435, the crash disappeared but there still are some exceptional outputs as below:
$ realsense-viewer 
 19/11 00:44:55,410 ERROR [140442209351232] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 19/11 00:44:55,410 WARNING [140442367862336] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Enable / disable backlight compensation
 19/11 00:44:55,410 WARNING [140442367862336] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image brightness
 19/11 00:44:55,410 WARNING [140442367862336] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image contrast
 19/11 00:44:55,410 WARNING [140442367862336] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Controls exposure time of color camera. Setting any value will disable auto exposure
 19/11 00:44:55,410 WARNING [140442367862336] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image gamma setting
 19/11 00:44:55,410 WARNING [140442367862336] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image hue
 19/11 00:44:55,410 WARNING [140442367862336] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image saturation setting
 19/11 00:44:55,410 WARNING [140442367862336] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image sharpness setting
 19/11 00:44:55,410 WARNING [140442367862336] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
 19/11 00:44:55,410 WARNING [140442367862336] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Enable / disable auto-exposure
 19/11 00:44:55,410 WARNING [140442367862336] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Enable / disable auto-white-balance

here maybe a normal operation flow: **first start it without D435, then connect/plugin D435**

BR.

MartyG-RealSense, 2020-11-19:
If the above errors occur, is the camera still detected in the side-panel of the Viewer and are you able to activate any streams successfully?

viisautta, 2020-11-20:
If error occur, this gui flashes and quit.
Only after firstly starting it without D435 and then connecting/plugining it, the gui is ok and any streams can be activated normally.

Thanks:)

MartyG-RealSense, 2020-11-20:
It is great to hear that you can stream successfully.  Do you need any further assistance please?  Thanks!

viisautta, 2020-11-20:
That's ok. thanks.

closed 2020-11-20 by viisautta

mintnguyenn, 2021-12-08:
Hi @MartyG-RealSense, I have an error that "failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS" when running realsense-viewer. How can I solve this problem? Thanks

MartyG-RealSense, 2021-12-08:
Hi @mintnguyenn   Could you provide information about the questions below, please.

-  Which RealSense camera model are you using?

-  Which computer or computing device are you using, please?  

-  Which method did you use to install the librealsense SDK?

-  Does the error still occur if you launch the Viewer without a camera attached and then insert the camera into the USB port after the Viewer has opened?

mintnguyenn, 2021-12-09:
Hi @MartyG-RealSense,
- I am using D435.
- I am using HP Omen 15 2020, with AMD processor.
- I installed the SDK with "sudo apt-get install librealsense2-dkms".

MartyG-RealSense, 2021-12-09:
Thanks very much for the information.  In addition to installing 'librealsense2-dkms', did you also install support for SDK examples and tools with **sudo apt-get install librealsense2-utils** please?

[image]

mintnguyenn, 2021-12-09:
Hi @MartyG-RealSense , yes I have installed everything in the description. And there is 1 problem is that maybe my kernel (which is 5.11.0-41 is not suitable with the SDK version. Therefore when I run 'modinfo uvcvideo | grep "version:"' it does not include realsense string. Is it the reason lead to this error? 

When I use the camera via ros (the ros wrapper), the rviz can capture the streaming very well, so I think the problem mainly come from the SDK version

MartyG-RealSense, 2021-12-09:
Hi @mintnguyenn  The librealsense SDK does not officially support kernel 5.11 yet.   At the time of writing this, the most recent supported kernel version is 5.8.  It is possible to use kernels newer than the officially supported ones but the results may be unpredictable in regards to stability.

If librealsense is built from source code with the **RSUSB backend** method then it is not dependent on Linux versions or kernel versions and does not require patching.