realsenseai/librealsense #7798: Realsense D435 error: FrameSync not supported

viisautta, 2020-11-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.08.200| 
| Operating System & Version |  Fedora33 | 
| Kernel Version (Linux Only)    | 5.9.8-200.fc33.x86_64 | 
| Platform                                 | laptop |
| SDK Version                            |  2.39.0  | 
| Language                            | C++| 

### Issue Description
$ pcl_openni2_viewer 
[RS2] +Rs2Driver
[RS2] initialize 
[RS2] enumerateDevices 
[RS2] devicesChanged removed=(nil) added=0x1ed5ef0
[RS2] deviceConnected
[RS2] queryDeviceInfo 
[RS2] DEVICE serial=750612070937 name=Intel RealSense D435
[RS2] Rs2Driver INITIALIZED
[RS2] deviceOpen uri=750612070937
[RS2] +Rs2Device
[RS2] initialize 
[RS2] queryDeviceInfo 
[RS2] DEVICE serial=750612070937 name=Intel RealSense D435
[RS2] initializeStreams 
[RS2] SENSOR 0
...
[RS2] updateConfiguration 
**[RS2] ERROR at FILE /home/about/Public/Devel/librealsense-2.39.0/wrappers/openni2/src/Rs2Driver.cpp LINE 258 FUNC enableFrameSync
	FrameSync not supported
void pcl::io::openni2::OpenNI2Device::setSynchronization(bool) @ /home/about/Public/Devel/pcl-1.11.1/io/src/openni2/openni2_device.cpp @ 428 : Enabling depth color synchronization failed:** 

[RS2] setProperty propertyId=5 dataSize=4
...

MartyG-RealSense, 2020-11-18:
Hi @viisautta   I note from the top of your kindly provided log that you are using **OpenNI2**.  There has been a few previous cases of **FrameSync not supported** when using **enableFrameSync** with OpenNI2 and D435 / D435i.  

https://github.com/IntelRealSense/librealsense/issues/5001
https://github.com/IntelRealSense/librealsense/issues/5227
https://github.com/IntelRealSense/librealsense/issues/6441

It seems that the error is sourced from the line linked to below in the **Rs2Driver.cpp** file of the librealsense OpenNI2 wrapper.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/openni2/src/Rs2Driver.cpp#L256

There is no successful outcome to any of these cases, unfortunately.

viisautta, 2020-11-18:
thanks for you tips.
it seems a feature unimplemented up to now.

MartyG-RealSense, 2020-11-19:
I will enquire to Intel about it to seek feedback.

viisautta, 2020-11-20:
thanks a lot, looking forward to completion of this driver. 
:)

MartyG-RealSense, 2020-11-26:
Hi @viisautta   I am still awaiting feedback from Intel about FrameSync and have followed up on it.  Thanks very much for your patience!

MartyG-RealSense, 2021-11-16:
Closed due to decision not to progress with implementing the requested feature.

closed 2021-11-16 by MartyG-RealSense