realsenseai/librealsense #7799: librealsense::align and librealsense::align_sse give different outputs for depth reprojection

mhorowitzgelb, 2020-11-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | (05.12.07.100) | 
| Operating System & Version |   {gLinux)  | 
| Kernel Version (Linux Only)    |  (5.7.17-1)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.3.8 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {other }                          | 
| Cpu       Intel© Xeon© W-2135 CPU @ 3.70GHz × 6
### Issue Description
For the longest time I couldn't figure out why but using the sse_version of align gives different alignment than unoptimized version. The Unoptimized version gives the correct alignment. 

For more context. I just upgraded the firmware to the one specified above using the Windows firmware tool. I also recalibrated and set the extrinsics using the windows tool.

[image]
[image]

mhorowitzgelb, 2020-11-18:
[image]
here is the color image that the other two should be aligned to. You can see the librealsense::align the second photo is properly aligned hoewever the first is not.

MartyG-RealSense, 2020-11-18:
Hi @mhorowitzgelb   My understanding of the align processing block is that it uses SSE optimizations if the computer's hardware supports it, otherwise it uses unoptimized align.  If CUDA support is enabled in librealsense on devices equipped with an Nvidia GPU then CUDA is used for alignment instead.

It seems that librealsense is using **SSSE3**.  Whilst this instruction set is supported by common Intel processor ranges such as Intel Core and Celeron, the range of Intel Xeon processors supported is somewhat limited.  So I wonder whether you are getting non-ideal results from SSE because you have a Xeon **W-2135** processor.

mhorowitzgelb, 2020-11-18:
Possibly. Though my colleague has the same cpu and it seems not to be an issue for him.

MartyG-RealSense, 2020-11-18:
There was a similar sounding past case where two colleagues were using the same model of PC, but one worked fine and the other had problems.  Eventually it was found that the colleague whose PC worked had updated Ubuntu on his computer and so had more recent drivers.  Once the person with the problem updated their computer too to match the confguration of the colleague's computer then the problem disappeared.

closed 2020-12-03 by MartyG-RealSense