realsenseai/librealsense #7800: converting .ply to obj - any solutions?

AndreV84, 2020-11-18:
```
python3 ply2obj.py cloud.ply cloud.obj
Traceback (most recent call last):
  File "ply2obj.py", line 52, in <module>
    main()
  File "ply2obj.py", line 16, in main
    convert(PlyData.read(args.ply_filename), args.obj_filename)
  File "ply2obj.py", line 35, in convert
    n = [ v['nx'], v['ny'], v['nz'] ]
ValueError: no field of name nx
```

used https://gist.github.com/Helias/268d48e8a925b6681fb7a6afe8169593

also tried pptk commands
```
>>> import pptk
>>> import numpy as np
>>> import plyfile
Note

If pip install plyfile does not work, simply save a local copy of plyfile.py from plyfile’s github page.

Read vertices in Truck.ply.

>>> data = plyfile.PlyData.read('Truck.ply')['vertex']
Use per-vertex attributes to make numpy arrays xyz, rgb, and n.

>>> xyz = np.c_[data['x'], data['y'], data['z']]
>>> rgb = np.c_[data['red'], data['green'], data['blue']]
>>> n = np.c_[data['nx'], data['ny'], data['nz']]
````

AndreV84, 2020-11-18:
also not able to visualize ply in  ctmviewer

MartyG-RealSense, 2020-11-18:
Hi @AndreV84   If you are able to make use of the Python point cloud library **pyntcloud**, perhaps you could use it to convert a ply file to a solid mesh by computing a **convex hull** with the guide in the link below:

https://stackoverflow.com/questions/44997054/point-cloud-to-volume

pyntcloud can read and write ply and obj formats:

https://pyntcloud.readthedocs.io/en/latest/io.html#i-o

AndreV84, 2020-11-18:
Thank you very much
Trying it

AndreV84, 2020-11-18:
```
python3
Python 3.6.9 (default, Oct  8 2020, 12:12:24) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from pyntcloud import PyntCloud
>>> diamond = PyntCloud.from_file("cloud.ply")
>>> convex_hull_id = diamond.add_structure("convex_hull")
>>> convex_hull = diamond.structures[convex_hull_id]
>>> diamond.to_file("diamond_hull.obj", also_save=["mesh"])
```
like that? or I need to specify actually custom [convex_hull_id] in the line 
convex_hull = diamond.structures[convex_hull_id] ?

AndreV84, 2020-11-18:
[image]
colors seem missed

[image]
[image]

input file
https://storage.googleapis.com/nvidia-dev/113620.ply

AndreV84, 2020-11-18:
@MartyG-RealSense 
from maintainers of the proposed tool:

Hola @AndreV84 , I'm afraid that pyntcloud does not currently support reading/writing to/from .obj files with per-vertex color information.

As far as I remember when adding the .obj code, vertex coloring is not considered part of the format "standard" and it is only supported by a few applications (i.e. Meshlab).

Looking at how it appears to be implemented in Meshlab (https://gamedev.stackexchange.com/a/66270) it might be feasible to add support to vertex coloring for the .obj format in pyntcloud with minor modifications, will change the label to Feature Request. Contributions are allways wellcome.

AndreV84, 2020-11-19:
meshlab seems to do the export\
also https://github.com/daavoo/pyntcloud/issues/293

closed 2020-11-19 by AndreV84

MartyG-RealSense, 2020-11-19:
Apologies for the delay in responding further.   Great news that you arrived at a solution!