realsenseai/librealsense #7806: How to calculate depth accuracy of D435 and D455?

JANGSOONMYUN, 2020-11-18:
In the datasheet, the depth accuracy of D435 at 2 meter working distance is less than 2%. And D455's depth accuracy at 4 meter is less than 2%.

So, if there is an object 1 meter away, depth accuracy is 20mm? But in the specification, it is normally 2~5mm depth accuracy at 1 meter working distance.

How can I calculate right depth accuracy?

wodndb, 2020-11-18:
@JANGSOONMYUN 
The error rate of depth "2%" is lough. There are not exist about absolute depth accuracy of D4xx series. Depth error of RealSense can be changed by light source, shape of object, and so on. So I recommend to measure error of RealSense by setting your camera environment and application.

MartyG-RealSense, 2020-11-18:
Thanks @wodndb for your advice to @JANGSOONMYUN !

Hi @JANGSOONMYUN  The above advice of @wodndb is right to say that accuracy can be affected by a range of factors, such as depth resolution, distance of the camera from the object and environmental conditions.

I would clarify that the percentage represents % of **the distance from the camera**.

Yes, at 1 meter distance an accuracy of 2.5 mm to 5 mm is expected with 400 Series cameras.

D455 has 2x the accuracy over distance of the D435 models.  So at 6 meters distance, the D455 should have the same depth accuracy as the D435 models do at 3 meters.  

You can get feedback about your camera's current depth accuracy using the **Depth Quality Tool** program.

JANGSOONMYUN, 2020-11-19:
Thank you @wodndb and @MartyG-RealSense 

However, you said "the percentage represents % of the distance from the camera". It means if the distance from the camera is 1 meter, then 2% of the distance is 20 mm. But why it is 2.5mm to 5mm in the datasheet? It is too different.

I understood accuracy can be affected by environments. I just want to know rough accuracy not exact accuracy.

wodndb, 2020-11-19:
@MartyG-RealSense 

Can we roughly calculate depth accuracy like this?

### D435
* 0 ~ 1 m depth depth error : 2.5 ~ 5 mm
   * Interpolate error value based distance.
   * ex: 0.8 meter --> depth error ≤ (2.5 + 0.8 × 2.5 = 4.5 ㎜)
* 1 ~2 m depth accuracy : 2% of distance.
   * Multiply percent to distance.
   * ex: 1.5 meter --> depth error ≤ (1500 × 0.02 = 30 ㎜)

MartyG-RealSense, 2020-11-19:
@JANGSOONMYUN   The full version of the statement is "less than 1% of the distance from the camera".  The camera is officially capable of accuracy of < 1%, though in real-world conditions 2% may be more easily achievable.  Even then though, there is no published explanation as far as I know for how the quoted 2.5 to 5 mm range of values is arrived at, so I cannot help with an explanation for that.

The official formula for calculating error over distance (RMS Error) is provided in point 5 of the section of Intel's camera tuning guide linked to below.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-verify-performance-regularly-on-a-flat-wall-or-target

[image]

JANGSOONMYUN, 2020-11-24:
Thank you very much @MartyG-RealSense @wodndb

closed 2020-11-24 by JANGSOONMYUN