realsenseai/librealsense #7810: Missing librealsense2-dkms package on focal

stfuchs, 2020-11-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | n/a | 
| Operating System & Version |  Linux (Ubuntu 20.04) | 
| Kernel Version (Linux Only)    |  5.4.0-48-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.40                           | 
| Language                            |  n/a                          | 
| Segment			|  Robot                          | 

### Issue Description
I was excited to see today's release for 20.04 support and tried to install the kernel patch module following these steps:

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages

It looks like the dkms package is still missing though:
```
$ apt-cache pkgnames | grep realsense
librealsense2-dbg
librealsense2-dev
librealsense2
librealsense2-gl-dbg
librealsense2-gl-dev
librealsense2-net
librealsense2-gl
librealsense2-utils
librealsense2-udev-rules
librealsense2-net-dbg
librealsense2-net-dev
```
Other packages are available:
```
$ apt-cache policy librealsense2
librealsense2:
  Installed: (none)
  Candidate: 2.40.0-0~realsense0.3815
  Version table:
     2.40.0-0~realsense0.3815 500
        500 http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo focal/main amd64 Packages
```

ev-mp, 2020-11-19:
@stfuchs , thanks for reporting this.
A synchronization issue with the server was identified and fixed, and the DKMS package is available for download now. 
Please check and update.

stfuchs, 2020-11-19:
@ev-mp thanks for the quick support. I managed to install it.

closed 2020-11-19 by stfuchs