realsenseai/librealsense #7813: [Question] How do you know when the .bag file has finished recording?

FLS-BP-US, 2020-11-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L500 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10)  | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.40.0.2482 }                          | 
| Language                            |  {python}                          | 
| Segment			|  {pyrealsense2 (2.40.0.2483) }                          | 

How do I know when the bag file has finished saving so the pipeline can be stopped?  Also, how do I use poll_for_frames so I can pause/resume the recorder or playback without the timeout? I cannot translate the c++ example to python and any other example I have found only shows the first frame.  (If you would like to use my code as an example for tkinter and realsense you can).

from tkinter import Frame
import cv2
import numpy as np
import pyrealsense2 as rs
from PIL import Image, ImageTk
import tkinter as tk
import tkinter.filedialog as tkFileDialog
from tkinter.constants import LEFT, TRUE
from tkinter import ttk
from time import sleep

class Application:
    def __init__(self):
        self.window = tk.Tk()
        self.window['bg'] = '#002A54'
        self.window.title("tkinter + pyrealsense playback")
        self.window.protocol('WM_DELETE_WINDOW', self.destructor)

        self.combo = ttk.Combobox(self.window, values=["Color","Depth"], state="readonly", postcommand=self.combo_changed)
        self.combo.current(0)
        self.combo.pack(side="top", fill="x", padx="5", pady="5")

        self.mainImage = tk.Label(master=self.window)
        self.mainImage.pack(side="top", fill="both", expand="yes", padx="5", pady="5")

        self.fr_buttons = tk.Frame(self.window, bg=self.window["bg"])
        self.fr_buttons.pack(side="bottom", fill="x")
        openButton = tk.Button(self.fr_buttons, text="Open", padx=10, pady=5, command=self.play_bag)        
        openButton.pack(side="left", fill="x", expand="yes", padx="5", pady="5")
        stopButton = tk.Button(self.fr_buttons, text="Stop", padx=10, pady=5, command=self.stop_bag)        
        stopButton.pack(side="left", fill="x", expand="yes", padx="5", pady="5")
        self.playing = False
        self.pipeline = None
        self.playback = None

    def destructor(self):
        """ Destroy the root object and release all resources """
        print("[INFO] closing...")
        if self.pipeline is not None:
            self.playing = False
            sleep(0.1)
            self.pipeline.stop()
            self.pipeline = None
        self.window.destroy()
        cv2.destroyAllWindows()  # it is not mandatory in this application
    
    def combo_changed(self):
        self.combo.update()
        
    def play_bag(self):
        if self.pipeline is None:
            self.pipeline = rs.pipeline()
        else:
            self.pipeline.stop()
            self.pipeline = rs.pipeline()
        path = tkFileDialog.askopenfilename(initialdir="C:/rosbag/", filetypes = (("bag files","*.bag"),("all files","*.*")))
        try:
            # Config pipeline
            config = rs.config()
            rs.config.enable_device_from_file(config, path, repeat_playback=False)
            config.enable_stream(rs.stream.depth)
            config.enable_stream(rs.stream.color)
            profile = self.pipeline.start(config)
          
            colorizer = rs.colorizer()

            self.playback = profile.get_device().as_playback()
            self.playback.set_real_time(TRUE)

            n = 0
            #Thread sleep so the stream has time to start playing
            self.playing = True
            sleep(0.1)           
            while (self.playing):
                n += 1
                frames=self.pipeline.wait_for_frames()

                if self.combo.current() == 1:
                    # Just Depth frame
                    depth_frame = frames.get_depth_frame()
                    if not depth_frame: continue
                    depth_color_frame = colorizer.colorize(depth_frame)
                    color_image = np.asanyarray(depth_color_frame.get_data())
                else:
                    #if self.combo.current() == "Color":
                    # Just Color frame
                    color_frame = frames.get_color_frame()
                    if not color_frame: continue
                    color_image = np.asanyarray(color_frame.get_data())
                
                img = Image.fromarray(color_image)
                imgtk = ImageTk.PhotoImage(image=img)
                self.mainImage.imgtk = imgtk
                self.mainImage.configure(image=imgtk)
                self.mainImage.update()
        except:
            self.mainImage.configure(text="End Of File")            
        finally:
            pass

    def stop_bag(self):
        if self.pipeline is not None:
            self.playing = False
            sleep(0.1)
            self.pipeline.stop()
            self.pipeline = None

print("[INFO] starting...")
pba = Application()
pba.window.mainloop()

RealSenseSupport, 2020-11-30:
@FLS-BP-US Please refer to https://github.com/IntelRealSense/librealsense/blob/master/examples/record-playback/rs-record-playback.cpp#L126 to add the control of stop recording or pause recording. Thanks!

FLS-BP-US, 2020-11-30:
I looked at that example before, it is in c++ and it does not use wait_for_frames or poll_from_frames.  My final project requires me to analyze the frames for circles or lines, while I am recording, streaming, or playing back.  I found the pyglet example and I will try using a timer in combination to try_wait_for_frames.  This should resolve the pausing issue.  Is there a pipeline notification or status about saving of the bag file? (If I do not wait more than 5 seconds between ending the recording and calling pipeline.stop() then the bag files has not saved completely and cannot be opened).

FLS-BP-US, 2020-12-01:
The following code works to record a bag file, but the pause() method.  I can pause the playback using this approach, is there a different way to pause a recorder?

```
from tkinter import Frame
from threading import Timer
import cv2
import numpy as np
import pyrealsense2 as rs
from PIL import Image, ImageTk
import tkinter as tk
import tkinter.filedialog as tkFileDialog
from tkinter.constants import LEFT, TRUE
from tkinter import ttk
from time import sleep
from pathlib import Path

class InfiniteTimer():
    """A Timer class that does not stop, unless you want it to."""

    def __init__(self, seconds, target):
        self._should_continue = False
        self.is_running = False
        self.seconds = seconds
        self.target = target
        self.thread = None

    def _handle_target(self):
        self.is_running = True
        self.target()
        self.is_running = False
        self._start_timer()

    def _start_timer(self):
        if self._should_continue: # Code could have been running when cancel was called.
            self.thread = Timer(self.seconds, self._handle_target)
            self.thread.start()

    def start(self):
        if not self._should_continue and not self.is_running:
            self._should_continue = True
            self._start_timer()
        else:
            print("Timer already started or running, please wait if you're restarting.")

    def cancel(self):
        if self.thread is not None:
            self._should_continue = False # Just in case thread is running and cancel fails.
            self.thread.cancel()
        else:
            print("Timer never started or failed to initialize.")

class Application:
    def __init__(self):
        self.window = tk.Tk()
        self.window['bg'] = '#002A54'
        self.window.title("tkinter + pyrealsense recorder")
        self.window.protocol('WM_DELETE_WINDOW', self.destructor)

        self.combo = ttk.Combobox(self.window, values=["Color","Depth"], state="readonly", postcommand=self.combo_changed)
        self.combo.current(0)
        self.combo.pack(side="top", fill="x", padx="5", pady="5")

        self.mainImage = tk.Label(master=self.window)
        self.mainImage.pack(side="top", fill="both", expand="yes", padx="5", pady="5")

        self.fr_buttons = tk.Frame(self.window, bg=self.window["bg"])
        self.fr_buttons.pack(side="bottom", fill="x")
        openButton = tk.Button(self.fr_buttons, text="Record", padx=10, pady=5, command=self.record_bag)        
        openButton.pack(side="left", fill="x", expand="yes", padx="5", pady="5")
        pauseButton = tk.Button(self.fr_buttons, text="Pause/Resume", padx=10, pady=5, command=self.pause)        
[... 74 lines cut ...]
        profile = self.pipeline.get_active_profile()
        device = profile.get_device()
        recorder = device.as_recorder()
        if self.paused:
            self.paused = False
            recorder.resume()
        else:
            self.paused = True
            recorder.resume()

    def stop(self):
        if self.pipeline is not None:
            if self.recording:
                self.recording = False
                if self.t is not None:
                    self.t.cancel()
                    sleep(10)
                    self.t = None
            self.pipeline.stop()
            self.pipeline = None

print("[INFO] starting...")
pba = Application()
pba.window.mainloop()      
```

FLS-BP-US, 2020-12-01:
I got the pause resume functionality to work with the following code.

```
from tkinter import Frame
from threading import Timer
import cv2
import numpy as np
import pyrealsense2 as rs
from PIL import Image, ImageTk
import tkinter as tk
import tkinter.filedialog as tkFileDialog
from tkinter.constants import LEFT, TRUE
from tkinter import ttk
from time import sleep
from pathlib import Path

class InfiniteTimer():
    """A Timer class that does not stop, unless you want it to."""

    def __init__(self, seconds, target):
        self._should_continue = False
        self.is_running = False
        self.seconds = seconds
        self.target = target
        self.thread = None

    def _handle_target(self):
        self.is_running = True
        self.target()
        self.is_running = False
        self._start_timer()

    def _start_timer(self):
        if self._should_continue: # Code could have been running when cancel was called.
            self.thread = Timer(self.seconds, self._handle_target)
            self.thread.start()

    def start(self):
        if not self._should_continue and not self.is_running:
            self._should_continue = True
            self._start_timer()
        else:
            print("Timer already started or running, please wait if you're restarting.")

    def cancel(self):
        if self.thread is not None:
            self._should_continue = False # Just in case thread is running and cancel fails.
            self.thread.cancel()
        else:
            print("Timer never started or failed to initialize.")

class Application:
    def __init__(self):
        self.window = tk.Tk()
        self.window['bg'] = '#002A54'
        self.window.title("tkinter + pyrealsense recorder")
        self.window.protocol('WM_DELETE_WINDOW', self.destructor)

        self.combo = ttk.Combobox(self.window, values=["Color","Depth"], state="readonly", postcommand=self.combo_changed)
        self.combo.current(0)
        self.combo.pack(side="top", fill="x", padx="5", pady="5")

        self.mainImage = tk.Label(master=self.window)
        self.mainImage.pack(side="top", fill="both", expand="yes", padx="5", pady="5")

        self.fr_buttons = tk.Frame(self.window, bg=self.window["bg"])
        self.fr_buttons.pack(side="bottom", fill="x")
        openButton = tk.Button(self.fr_buttons, text="Record", padx=10, pady=5, command=self.record_bag)        
        openButton.pack(side="left", fill="x", expand="yes", padx="5", pady="5")
        pauseButton = tk.Button(self.fr_buttons, text="Pause/Resume", padx=10, pady=5, command=self.pause)        
[... 74 lines cut ...]
        profile = self.pipeline.get_active_profile()
        device = profile.get_device()
        recorder = device.as_recorder()
        if self.paused:
            self.paused = False
            recorder.resume()
        else:
            self.paused = True
            recorder.pause()

    def stop(self):
        if self.pipeline is not None:
            if self.recording:
                self.recording = False
                if self.t is not None:
                    self.t.cancel()
                    sleep(10)
                    self.t = None
            self.pipeline.stop()
            self.pipeline = None

print("[INFO] starting...")
pba = Application()
pba.window.mainloop() 
```

RealSenseSupport, 2020-12-02:
@FLS-BP-US I don't think there's pipeline notification or status about saving of the bag file. Do you mean the pause resume function resolves your issue?

FLS-BP-US, 2020-12-02:
I can pause and resume.  Can I request you guys add a pipeline status like: starting, stopping, running, stopped?

closed 2020-12-02 by FLS-BP-US