realsenseai/librealsense #7814: How to set the camera focal length !!!

hustwb, 2020-11-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D430| 
| Firmware Version                   | 05.12.01.00| 
| Operating System & Version |   Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                        | 
| Platform                                 | NVIDIA Jetson Nano|
| SDK Version                            | 2.36.0                      | 
| Language                            |  C++       | 
| Segment			|  others             | 

### Issue Description
there is my camera focal length :  intrinsics.fx:387.386, intrinsics.fy:387.386, and image size is 640(Width)*480(height)

I want to set the focus of the camera a little bigger,But I don't know which function to use?

MartyG-RealSense, 2020-11-19:
Hi @hustwb   I have an idea about what you are asking about, but just to make sure that we are both thinking of the same thing, can you tell me more please about what you are trying to achieve with a 'bigger focus'?  

For example, do you wish to zoom the image in / out?  Or do you want the image to be less blurred when moving the camera very close to a surface?

dorodnic, 2020-11-24:
Maybe I'm mistaken, but changing focal length in the calibration table would not have any effect on camera focus (@agrunnet ?)

agrunnet, 2020-11-24:
Correct. The focus of the lenses are fixed. You cannot change them in software.

MartyG-RealSense, 2020-12-08:
Case closed due to no further comments received.

closed 2020-12-08 by MartyG-RealSense

JohnVorwald, 2021-03-31:
When I enter the following command, the results appear to have several focal lengths.  Which of these are applicable to real sense 435i.  We are trying to see objects at distances > 40 m.

rs-enumerate-devices -c > calibr_coeff.txt

agrunnet, 2021-03-31:
Sorry you cannot set the focal length. This is a fixed focus lens.

HelloCentaurus, 2021-05-12:
@MartyG-RealSense  The QRcode in realsense RGB Image is two small,I wish to zoom the image in, but I don't have a good solution. What do you suggest？Tanks！

MartyG-RealSense, 2021-05-12:
Hi @HelloCentaurus  A similar subject about the challenges of reading barcodes from a distance with an RGB image is discussed in the link below.

https://github.com/IntelRealSense/librealsense/issues/7187

In that case, I provided advice about settings that could be adjusted to sharpen the RGB image to make the code more readable.

https://github.com/IntelRealSense/librealsense/issues/7187#issuecomment-690248295

[image]

If your project is able to use OpenCV then using the instruction **cv2.INTER_LINEAR** to zoom the RGB image may be a solution.

https://www.geeksforgeeks.org/image-resizing-using-opencv-python/

HelloCentaurus, 2021-05-14:
@MartyG-RealSense Thank you for your reply。I already searched the issues,and had found your other comments. So there is two way:Sharpen the image or Expand the size of the image.These are a way, but not a great way。But thanks,I  know the limits of the camera,and I will use other methods to replace QR code recognition.

MartyG-RealSense, 2021-05-14:
Thanks very much  @HelloCentaurus  - good luck!