realsenseai/librealsense #7815: Thw error shows when I connect the D435 with realsense viewer

SUN-GE, 2020-11-19:
This error shows when I connect the D435 with realsense viewer. Does this mean my device is broken?

Invalid Value in rs2_get_option(options:000001A64B57D8C0, option:Emitter On Off):
hwmon command 0x7d( 0 0 0 0 ) failed (response -21= No data to return)

MartyG-RealSense, 2020-11-19:
Hi @SUN-GE   My experience with this error is that it does not mean that the camera hardware is broken, as the camera may work fine if plugged into another computer.

Could you tell me wich computer or computing device you are connecting the D435 to, please?

SUN-GE, 2020-11-19:
Thanks for the quick response. I am using ubuntu 18.04. I connected D435 with my desktop.

SUN-GE, 2020-11-19:
I have checked it. you are right. It works at another computer.

SUN-GE, 2020-11-19:
Thanks for quick response. I am using ubuntu 18.04 (my desktop). I have check it, it works with another computer.

MartyG-RealSense, 2020-11-19:
Do you wish to explore why it has problems on the first computer you tried please, or are you satisifed to use the camera on the other computer?

dorodnic, 2020-11-24:
This happens when you are using new firmware (5.12.8.200+) with older SDK (<2.39) and mentioned in the release notes - https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#new-features-1
The warning itself is harmless, unless you want to use Emitter On-Off feature. If that is the case, you need compatible SDK-firmware pair.

MartyG-RealSense, 2020-12-01:
Hi @SUN-GE   Do you require further assistance with this case, please?   Thanks!

SUN-GE, 2020-12-02:
Thanks for your feedback! My problem is solved.

closed 2020-12-02 by SUN-GE

MartyG-RealSense, 2020-12-02:
I'm very pleased to hear of your success.  Thanks for the update!