realsenseai/librealsense #7816: how to reduce cpu in arm?

Johnlan197, 2020-11-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I am using d435i testing in arm,when cross-compile realsense sdk2.39,i have turn off build_with_openmp,but using sdk to run program in arm, cpu is about 77 prectange in single core(capture 640*480 rgb and depth in 30fps).Do you have any suggestion to reduce cpu?

MartyG-RealSense, 2020-11-19:
Hi @Johnlan197  The SDK's CMake build configuration documentation confirms that having **OpenMP** enabled can result in greater CPU utilization when performing YUY to RGB conversion if OpenMP = true.

https://github.com/IntelRealSense/librealsense/wiki/Build-Configuration

[image]

In regard to ways to reduce CPU usage:

On computers equipped with an NVidia GPU, CUDA support can be enabled to optimize YUY to RGB conversion, and also reduce the CPU usage when performing depth to color alignment.  This is because CUDA offloads some of the CPU's work onto the GPU.

If you are building librealsense with CMake, you can speed up building by using a '-j' term in the make instruction to define how many cores the build process can make use of.  This may cause random crashes when used on low-end devices though.

The librealsense documentation for building from source with this multicore recommends '-j8', though with high-end processors with a large number of cores you may be able to use as high as '-j16'.  The j value represents the number of processor **threads**, so -j8 is representative of a quad-core processor: 4 (cores) x 2 (parallel threads per core) = total number of threads.

[image]

You may also be able to reduce CPU usage by making use of GLSL Processing Blocks when programming SDK scripts.  GLSL is an alternative to CUDA that also offloads CPU work to the GPU, and has the advantage that it is **vendor-neutral** (it should work with any GPU, though may not produce noticable improvement on low-end devices).

The link below has a very good pros and cons analysis of GLSL processing blocks and when it can be used.

https://github.com/IntelRealSense/librealsense/pull/3654

Adding GL support is straightforward, as it is usually a matter of putting **GL::** int front of instructions that start with **RS2::**.  For example:

**rs2::pointcloud,**
becomes
**rs2::gl::pointcloud**

Johnlan197, 2020-11-19:
@MartyG-RealSense Thank you for reply, i have some questions
1. i have test realsense by AMCap tool, this tool can capture realsense data,the cpu cost is very low compared to use realsense sdk to capture, why using sdk cost more cpu?

MartyG-RealSense, 2020-11-19:
The percentage of CPU usage in the SDK will be influenced by the operations being performed by a particular program.  For example, CPU usage may rise if depth to color alignment is being performed and the processing is not being offloaded from the CPU onto a graphics GPU.  So for some programs CPU usage wll be low and for some it will be high, depending on what the program is doing.  High CPU usage may also be caused by inefficiencies in a program.

closed 2020-12-03 by MartyG-RealSense