realsenseai/librealsense #7817: Changing depth unit won't take effect immediately

Domos, 2020-11-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.12.7.100| 
| Operating System & Version |  Debian  | 
| Kernel Version (Linux Only)    |  5.7.17                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.38.1                          | 
| Language                            |  C                          | 
| Segment			|  others                        | 

### Issue Description
After changing depth unit to 0.0001, `pipeline_.wait_for_frames()` still produces frames with 0.001 (default) depth unit for a while.
We found it by looking into the depth values, and also alignment won't work.
`frameset.get_depth_frame().get_units()` says 0.0001 but the frame data is still 0.001.
We suspect frames are queued so we still receives old frames after changing the depth unit.
Or it takes a while for realsense to switch the depth unit.
The pipeline model does not allow us to set depth unit before streaming.
Any idea?

```
config cfg;
cfg.enable_stream(rs2_stream::RS2_STREAM_DEPTH);
pipeline_profile profile = pipeline.start(cfg); // starts to stream frames with 0.001 depth unit?
depth.set_option(RS2_OPTION_DEPTH_UNITS, 0.0001);
auto frameset = pipeline_.wait_for_frames(); // 0.001 depth unit
frameset.get_depth_frame().get_units(); // 0.001, which is wrong. also causing align to fail
frameset = pipeline_.wait_for_frames(); // 0.001 depth unit
frameset = pipeline_.wait_for_frames(); // 0.001 depth unit
...
frameset = pipeline_.wait_for_frames(); // finally we got a frame with 0.0001 depth unit
// but how do we know without looking into the values?
```

MartyG-RealSense, 2020-11-19:
Hi @Domos   Changing an option can take at least a couple of frames to complete, due to the time taken for changes to propogate back to the user.

If you are using C language as indicated in the details provided at the top of this case and not C++, the librealsense C wrapper has an example program with a line that checks depth scale.

https://github.com/IntelRealSense/librealsense/blob/master/examples/C/depth/rs-depth.c#L53

Chandler, 2020-11-19:
@Domos we experienced the same thing and decided the best thing was to just drop a few frames after a settings change. There is no way I know of to infer the depth unit from the frame itself unless you want to use a heuristic about the depth values themselves and their expected value.

Domos, 2020-11-19:
Hi,
We decided to set depth unit earlier and drop a few frames as @Chandler mentioned.
That did work.
Thanks!

```
device.query_sensors()
depth.set_option(RS2_OPTION_DEPTH_UNITS, 0.0001);

config cfg;
cfg.enable_stream(rs2_stream::RS2_STREAM_DEPTH);
pipeline_profile profile = pipeline.start(cfg); // starts to stream frames with 0.001 depth unit?
// drop 30 frames
for (int i = 0; i < 30; ++i) {
  pipeline_.wait_for_frames();
}
auto frameset = pipeline_.wait_for_frames(); // 0.0001 depth unit
```

closed 2020-11-19 by Domos

MartyG-RealSense, 2020-11-20:
Thanks again for your assistance @Chandler 

@Domos   I'm pleased to hear that you achieved a solution that is satisfactory for you.  Thanks for the update!