realsenseai/librealsense #7819: What causes get_frame_number() to reset?

stfuchs, 2020-11-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.03.00 | 
| Operating System & Version |   Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)    |  4.15.0-1100-oem                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.33.1                          | 
| Language                            |  C                           | 
| Segment			|  Robot                          | 

### Issue Description
I have a question regarding the frame number assigned to each frame: 

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1frame.html#ac937e74ea40e08cd5ca46fe0ad302939

Since it is set by the device part of the frame meta data I figured I could use it to identify dropped frames by checking its increments. However I have noticed that occasionally the number resets and starts back at 0. Is this intentional and if so what is causing it?

As a side note: I'm using 8 cameras connected via to a usb3 hub. All cameras are streaming depth + left IR both at 848x480 and 6 Hz.

MartyG-RealSense, 2020-11-19:
Hi @stfuchs   Are your multiple cameras hardware synced and connected together with physical sync cabling, please?

stfuchs, 2020-11-19:
@MartyG-RealSense no they are not.

MartyG-RealSense, 2020-11-19:
Is there a possibility that the hub USB is cutting out and disconnecting all cameras, and then automatically reconnecting to resume streaming?  My recollection is that if reconnection ocurs within 5 seconds of disconnection then the pipeline automatically resumes.

MartyG-RealSense, 2020-11-19:
The link below sounds similar to your situation actually, except that in that case it is depth + RGB streaming at the same FPS that causes the reset to 0 instead of depth + left IR.

https://github.com/IntelRealSense/librealsense/issues/6130#issuecomment-603982335

stfuchs, 2020-11-19:
Thanks Marty. I'm not using rgb and the resets happened randomly without me interfering with the cameras.

I have the suspicion too that the usb hub might be cutting out power for a few milliseconds. However it is never all cameras at once, usually just a single one that resets the counter. Occasionally there are frame drops involved too:

``` 
 19/11 11:17:59,926 INFO [139655869613824] (types.h:1051) Incomplete video frame detected!
Size 427008 out of 814335 bytes (52%)
```
and I also see
```
 19/11 11:28:30,634 ERROR [139656154834688] (ds5-options.cpp:86) Asic Temperature value is not valid!
```
from time to time. 

I'm just tying to confirm if it's possible for the camera to disconnect somewhat unnoticed on short power outages. On the other hand when I manually unplug the camera and quickly (<1s) plug it back in the behavior is a lot different and there are errors all over the place:
```
 19/11 11:36:06,265 ERROR [139653982172928] (types.h:307) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device
 19/11 11:36:06,265 INFO [139653982172928] (frame-archive.h:213) The user was holding on to 1 frames after stream 0x0x7f03ac2322a0 stopped
 19/11 11:36:06,265 INFO [139653982172928] (frame-archive.h:213) The user was holding on to 1 frames after stream 0x0x7f03ac281940 stopped
 19/11 11:36:06,266 INFO [139653629843200] (backend-v4l2.cpp:889) V4L stream is closed
 19/11 11:36:06,266 ERROR [139653982172928] (types.h:307) xioctl(VIDIOC_STREAMOFF) failed for buf_type=1 Last Error: No such device
 19/11 11:36:06,266 INFO [139653982172928] (frame-archive.h:224) All frames from stream 0x0x7f03ac2322a0 are now released by the user
 19/11 11:36:06,266 INFO [139653982172928] (frame-archive.h:224) All frames from stream 0x0x7f03ac281940 are now released by the user
 19/11 11:36:06,287 ERROR [139653545981696] (types.h:307) xioctl(VIDIOC_DQBUF) failed for fd: 66 Last Error: No such device
 19/11 11:36:06,287 ERROR [139653545981696] (backend-v4l2.cpp:1327) xioctl(VIDIOC_DQBUF) failed for fd: 66 Last Error: No such device
 19/11 11:36:06,287 INFO [139653545981696] (types.h:1051) xioctl(VIDIOC_DQBUF) failed for fd: 66 Last Error: No such device
```

I guess my question is are there other reasons for the counter to reset then a power disconnect caused by the hub? Another one you already gave me is when the rgb color sync kicks in to restart the depth stream. I also found 

> Fix hardware fps calculation when the frame number being reset internally by the FW, such as in `SCP Overflow` case.

on https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2331. I don't know what `SCP Overflow` means though.

MartyG-RealSense, 2020-11-19:
Yes, there is scope for errors occurring with a rapid unplug-replug, or unplugging the camera whilst streaming is active without stopping the stream first.  Slow or partial insertion of the camera into the USB port instead of a firm quick insertion action can also cause errors, or mis-identification of the camera as being on a USB 2.1 connection when it is in a USB 3 port.

The link below may be helpful in regard to incomplete video frame and SCP Overflow errors:

https://support.intelrealsense.com/hc/en-us/community/posts/360033706613-USB-SCP-overflow

Doronhi the RealSene ROS wrapper developer comments below about the invalid ASIc temperature error:

https://github.com/IntelRealSense/realsense-ros/issues/894#issuecomment-527315705

In regard to other reasons why frame counters reset: they can reset in multicam settings if cameras are connected together with hardware sync cabling to synchronize their timestamps and an electrostatic discharge (ESD) event occurs.  You confirmed though that you are not using hardware sync cabling.

stfuchs, 2020-11-23:
Thanks Marty. I order another usb hub (ACASIS 10 Ports 48W USB 3.0 Data Hub, https://www.amazon.com/gp/product/B07Q3TYF15/) and I'm running some tests right now. Will update once I know more.

MartyG-RealSense, 2020-11-23:
Thanks very much @stfuchs for the update - I look forward to hearing from you.

dorodnic, 2020-11-24:
Hi @stfuchs 
Whenever the firmware is doing internal stream reset, frame number will reset to 0. This happens when:
1. There are too many frame drops, possibly related to SCP overflow but in other frame drops cases as well
2. When you add / remove colour stream with same FPS - hardware sync kicks in and causes a reset
3. When certain advanced mode registers are updated (for instance, Disparity Shift)

stfuchs, 2020-11-25:
Sorry for the delay but I think I got some more insights. So first of I replaced my usb hub with the new one and had all 8 cameras run for about 10h straight. I experienced the exact same thing with my old hub: Every other hour or so set of 1-15 frames on a single camera would drop (`Incomplete video frame detected!`), the stream would reset and it'll work fine again. I kept track of how often a reset happened per camera. Here is what I got:
```
cam0: xxx
cam1: 
cam2: xx
cam3: xx
cam4: xxxxx x
cam5: x
cam6: xxxxx
cam7: xx
```

Obviously cam1 stands out so I double-checked and it turns out it is running on FW 5.12.8.200 while all the others are on 5.12.3.0. At first I figured it might just be a coincidence so I upgraded all cameras to 5.12.8.200 and ran another 10h test.

First outcome was that none of the 8 cameras reset their frame counter, so it seems like something has changed here. BUT: I had severe problems now. Every 2-5 hours 3 of the cameras completed stop streaming depth data and eventually recovered after 15 min to 1h. This was not the same 3 cameras but always different ones.

Then I tried the same test again  with only 4 cams connected and they running fine for 12h+ now. I have never experienced this behavior with older firmware...

stfuchs, 2020-11-26:
Yup happened again after ~4h of streaming in realsense viewer:
[image]
The depth stream of 3 cameras stopped for about 15 min, the IR stream was still working though. I have not seen this behavior with older firmware versions.

MartyG-RealSense, 2020-11-26:
Apologies for the delay in responding further @stfuchs 

I read your test results carefully.  It reminded me of a multicam behavior from 2017  with the RealSense SR300 camera model, where multiple SR300 cameras were being used on a USB hub for long-run sessions.  The amount of time that a session lasted before problems occurred seemed to be proportional to the number of cameras.  A pair of cameras could stream for up to a few days, but several cameras reduced the error-free time period noticably, to around 12-16 hours.

I recall that in one particular case where this was occurring, the RealSense user's successful solution was to use an **industrial grade** USB hub.

stfuchs, 2020-11-26:
@MartyG-RealSense no worries, I didn't expect any response over thanksgiving anyway :D. 

Maybe I was very clear. I actually noticed 2 separate things: 
(1) as described originally where both of my usb hubs work fine with 8 cameras connected except for those instance when a few images (1-15) are dropped and the camera stream reset. That's a little unfortunate but I can work with it. In fact we have worked with this setup on multiple robots where the cameras run for days and sometimes weeks without interruption. 
(2) after upgrading the cameras to FW 5.12.8.200 the depth stream on some cameras would occasionally (after 2-5h) stop completely for several minutes and eventually recover. This never happened before on older firmware versions. Maybe I should have opened a separate ticket for this (like on zendesk). 

I will try again with 4 cameras on each hub but I highly doubt the second problem is usb hub related. It seems to me there is an issue with the newer firmware.

MartyG-RealSense, 2020-11-27:
I note that the specification box at the top of this case states that you are using librealsense 2.33.1.  The SDK release notes state that if you have installed firmware **5.12.8.200** in your cameras then you need to upgrade to librealsense **2.39.0** due to internal firmware changes.  An SDK upgrade would mean that you may also get the benefit of improvements to multicam support that were added in SDK 2.35.2.

If it is not practical to update your project from SDK 2.33.1 then using firmware 5.12.7.100 should enable you to avoid the 2.39.0 update requirement that comes with 5.12.8.200.

A third option is downgrading all your cameras back to firmware **5.12.3.0** if that driver has proven stable for you, since that firmware version or newer is a recommended firmware for SDK version 2.33.1.

stfuchs, 2020-11-27:
@MartyG-RealSense I'm aware of the requirement changes. I made sure that I used
```dpkg -l | grep realsense
ii  librealsense2:amd64                                   2.39.0-0~realsense0.3571                         amd64        Intel(R) RealSense(tm) Cross Platform API - runtime
ii  librealsense2-dkms                                    1.3.14-0ubuntu1                                  all          Modified kernel modules for librealsense2
ii  librealsense2-gl:amd64                                2.39.0-0~realsense0.3571                         amd64        Intel(R) RealSense(tm) - GLSL-enabled extensions
ii  librealsense2-net:amd64                               2.39.0-0~realsense0.3571                         amd64        Intel(R) RealSense(tm) - Network Bridge extension
ii  librealsense2-udev-rules:amd64                        2.39.0-0~realsense0.3571                         amd64        Intel(R) RealSense(tm) Camera Capture API - udev rules
ii  librealsense2-utils:amd64                             2.39.0-0~realsense0.3571                         amd64        Intel(R) RealSense(tm) Camera Capture API - utils and demos
```
when I upgraded to FW 5.12.8.200. The box at the top described the case when I used 2.33.1 with 5.12.3.0 (except for that one camera that accidentally slipped through because it was used for a different test earlier).

I ran another tested with 8 cameras split across both hubs and each hub connected to a different usb port on my pc. I can confirm that overall it seemed more stable but still not resolved. It took about 12h until the same issue occured.

I'm not planing on upgrading our robots at the moment due to this incident. It would be great if you guys at Intel could see if you can reproduce this behavior. For now I'm sticking with 5.12.3.

MartyG-RealSense, 2020-11-28:
@stfuchs   I do not have enough cameras at my location to replicate your setup.  One of the other RealSense team members who has access to the same camera model in sufficient quantities may be able to conduct such a test at a future date.

MartyG-RealSense, 2020-12-05:
Hi @stfuchs   Given that I cannot provide an estimate of when a RealSense team member may be able to carry out a multi-camera test with the quantity of cameras that you are using, do you wish to keep this case open please?   Thanks!

stfuchs, 2020-12-07:
@MartyG-RealSense I opened another ticket with your customer support. Thanks.

MartyG-RealSense, 2020-12-08:
Thanks very much @stfuchs - as you have opened another ticket, do you wish to keep this case here open?   Thanks!

stfuchs, 2020-12-08:
Can be close. Thanks

closed 2020-12-08 by stfuchs

MartyG-RealSense, 2020-12-09:
Thanks for the update @stfuchs !