realsenseai/librealsense #7820: compare D455 & D415 depth quality

xiebei, 2020-11-20:
Why does the D455 have so many holes compared to the D415? High Accuracy preset.
Is there any way to improve?  thanks.

D415
[image]

D455
[image]

agrunnet, 2020-11-20:
Please use 848x480 for d455. Also set depth setting to “dense” and try to reduce the laser power when looking at near objects. This should all help a lot.

MartyG-RealSense, 2020-11-20:
I second the advice of @agrunnet above and would add that if you need a balance between accuracy and image fill then I recommend trying the **Medium Density** preset.

The images below are taken with D455 at 848x480 resolution.

[image]

[image]

Below is the same scene on the **High Density** preset, which provides a bit more image fill.

[image]

More information about Visual Presets (also known as **json**) and their traits can be found in the link below.

https://dev.intelrealsense.com/docs/d400-series-visual-presets

You can also define your own custom configuration in the RealSense Viewer and export it as a json  configurationfile using an option near the top of the Viewer's side-panel.  This file can then be loaded back into the Viewer or loaded with program code into your own application.

[image]

xiebei, 2020-11-23:
ok, I will try it, thanks.

closed 2020-11-23 by xiebei