realsenseai/librealsense #7821: depth byte binary text and image from Realsense D415, how do i change it to a mesh/3d-model for display in website?

elyazinho, 2020-11-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D415 } | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  Smartphone/Web                    | 

### Issue Description
Hi i have the depth byte binary text file and picture from the D415 realsense camera,
How do i use this to create a mesh or 3d-model, it doesnt have to be 360 degrees, i just have to make it 3D from the front for example? Can i do this with these output files?

MartyG-RealSense, 2020-11-20:
Hi @elyazinho   Could you confirm which file format you want to convert please (ply, raw, bin, csv, etc).  Thanks!

elyazinho, 2020-11-24:
@MartyG-RealSense, is gITF possible? if not we could work with ply as well. The files from the realsense camera are depth byte binary text file and image , do you have a step-by step instruction on how to convert to such formats? we can do it with python or Node.js
Do you need the sample binary text file and it's accompanying image?

MartyG-RealSense, 2020-11-24:
Hi @elyazinho    I am not aware of past cases of librealsense data being used with **gITF** unfortunately.

If at all possible, I would strongly recommend using **ply** files as the basis of your conversion to mesh or 3D model, as such a conversion is well supported by librealsense and by 3D modelling software..

The librealsense SDK offers the ability to configure ply-file export to generate the ply with on / off settings such as normals, color and binary   A Python example of this export mechanism is in the link below.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py#L40

There are other Python ply export refererences, such as this one:

https://github.com/IntelRealSense/librealsense/issues/3009

If you are adding the file to the website at a later time after it has been captured though, you may find it easier to just use the built-in **3D** point cloud mode of the **RealSense Viewer** program, which has these same ply file export options integrated into it.

closed 2020-12-08 by MartyG-RealSense