realsenseai/librealsense #7822: Make install error "config.h: No such file" in Ubuntu 20.04 LTS

prantoran, 2020-11-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i | 
| Firmware Version                   | - | 
| Operating System & Version |    Linux (Ubuntu 20.04 LTS)  | 
| Kernel Version (Linux Only)    |  5.4.0-53-generic  | 
| Platform                                 | PC |
| SDK Version                            | ~ 2.40.0 | 
| Language                            |  C/C++ | 
| Segment			|  VR/AR  | 

### Issue Description

I was trying the approach that worked in [6368](https://github.com/IntelRealSense/librealsense/issues/6368#issuecomment-636316279) but I got the following error when I ran `sudo make install` stated in the steps for installing for [Ubuntu](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#building-librealsense2-sdk):

```
...
[  1%] Performing build step for 'libusb'
Scanning dependencies of target usb
[  9%] Building C object CMakeFiles/usb.dir/libusb/core.c.o
/.../librealsense/build/third-party/libusb/libusb/core.c:23:10: fatal error: config.h: No such file or directory
   23 | #include "config.h"
      |          ^~~~~~~~~~
compilation terminated.
make[5]: *** [CMakeFiles/usb.dir/build.make:63: CMakeFiles/usb.dir/libusb/core.c.o] Error 1
make[4]: *** [CMakeFiles/Makefile2:76: CMakeFiles/usb.dir/all] Error 2
make[3]: *** [Makefile:130: all] Error 2
make[2]: *** [CMakeFiles/libusb.dir/build.make:113: libusb-prefix/src/libusb-stamp/libusb-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:1062: CMakeFiles/libusb.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

Also, `sudo make uninstall` did not work, which gave the error:
```
CMake Warning at cmake_uninstall.cmake:2 (MESSAGE):
  Cannot find install manifest:
  "/home/prantoran/work/src/github.com/IntelRealSense/librealsense/build/install_manifest.txt"

-- Uninstall targets will be skipped
Built target uninstall
```

I ran the commands in the `.../librealsense/build` folder.

MartyG-RealSense, 2020-11-20:
Hi @prantoran   This error with **config.h**has  been encountered in a past case when trying to use Ubuntu 20.04.   A RealSense user posted his solution in that discussion (not tested with the new SDK 2.40.0):

https://github.com/IntelRealSense/librealsense/issues/6436#issuecomment-653942842

prantoran, 2020-11-21:
Thank you, the missing `config.h` error got fixed after installing `libusb`. I also installed other packages mentioned in the [instruction](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites):
```
 sudo apt-get install git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev 
```

Now, I got a glu-perspective error and I tried installing freeglut3 but it did not fix the problem:
```
sudo apt-get install freeglut3 freeglut3-dev
```

I executed the following for `glfw` [support](https://github.com/IntelRealSense/librealsense/issues/7360#issuecomment-693525921)
```
sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev at 
```

prantoran, 2020-11-21:
Since this is a new issue, I am closing this one

closed 2020-11-21 by prantoran

MartyG-RealSense, 2020-11-21:
Thanks very much for the update and the details of the methods that worked for you!

christian-rauch, 2022-11-03:
The crucial part is installing the `libusb-1.0-0-dev` package:
```sh
sudo apt install libusb-1.0-0-dev
```
The other packages do not matter for this specific issue.

The CMake project is checking if `libusb-1.0` is installed and otherwise will use its internal version of libusb. You can check if the library is installed via:
```sh
pkg-config --modversion libusb-1.0
```
If this returned something like `1.0.25`, then CMake should detect this as well.